prism-em/prismatic #2: 4D output should crop center

apryor6, 2017-10-31:
The current 4D output results in many 2D images, each of which includes the anti-aliasing aperture. The result is that much of the array is wasted space (zeros). Only the central region should be returned.

matkraj, 2017-11-01:
4D output also has swapped quadrants

matkraj, 2017-11-23:
I am not 100% sure how to do this, but here is a suggestion for change to crop and swap quadrants for 4D STEM - I am not sure this is not breaking any other data saves
https://github.com/matkraj/prismatic/commit/d7c0a98ce29f8664ef499a6aafc098b4b14a0b85#diff-d87e66171dfb2a89f41ff2713110710f

apryor6, 2017-11-23:
The quadrant swapping functionality is already in Prismatic in the function `fftshift2` within include/utility.h. Also, the Prismatic array class has a to_mrc method to write out the data, so all you should need to do is create a new Prismatic array, fftshift the large one, and then copy the center into the smaller array before writing

matkraj, 2017-11-23:
my code was changing toMRC_f method for this. Thanks for `fftshift2` suggestion

cophus, 2017-12-05:
I prefer the default FFT quadrant positions because the origin pixel is always at (0,0).  I agree though that at some point we should definitely change the 4D output to only save the portions inside the anti aliasing aperture.

apryor6, 2017-12-05:
I think the solution would be to have only inside the anti aliasing aperture saved always, and then to add an additional option that will shift the quadrants if the user desires

lerandc, 2019-07-03:
Outputs are now cropped and fftshifted by default.

closed 2019-07-03 by lerandc