prism-em/prismatic #4: Complex output wave support

apryor6, 2017-10-31:
A user has requested that the complex exit wave been available as an output. This could be achieved in the current framework by outputting separate arrays containing the real and complex components of the exit wave at each probe position.

thomasaarholt, 2017-11-13:
I'd add my interest to this!

cophus, 2017-12-05:
This is a pretty straightforward extension.  Typically in my Matlab implementation, when I am exporting complex arrays to say HDF5 I split the real and imaginary along the final dimension vector.  So for example 4D-STEM outputs become 5D arrays with dimensions:

[k_x   k_y   probe_x   probe_y   {real, imaginary}]

HDF5 supports compound datatypes, so theoretically we could define a complex type - but most other formats I am aware (such as MRC, tiff images, etc) don't do this I think.

closed 2021-06-16 by lerandc