prism-em/prismatic #5: 4D frozen phonon calculation question

matkraj, 2017-11-08:
I was just wondering if this block of a code means that frozen phonon is only done for 3D output?
https://github.com/prism-em/prismatic/blob/ce5fed79545266d72a2d6e6aa4974230ff9bb4d8/src/Multislice_entry.cpp#L41-L73

apryor6, 2017-11-08:
Multiple frozen phonons are calculated. The block of code of interest computes all remaining frozen phonons after the first and stores them as a running average. If only one frozen phonon is requested, there is no need to create this second copy for the running average, and this step is skipped. Note that the inner loop starts at fp_num=1, which is the second configuration since C++ is zero-based for indexing.

matkraj, 2017-11-08:
Cool! I asked for the following reason. If I select i.e. 5 configurations for frozen phonon for 4D STEM output, I get 5 files per probe position (their name end with '..FP_1.mrc', '..FP_2.mrc'). However, these files are exactly the same, which confuses me a bit. My file *xyz file is:

> STO 
> 3.905   3.905   3.905
> 38 0.00000000 0.00000000 0.00000000   1   0.08
> 22 1.95250000 1.95250000 1.95250000   1   0.08
> 8 0.00000000 1.95250000 1.95250000   1   0.08
> 8 1.95250000 0.00000000 1.95250000   1   0.08
> 8 1.95250000 1.95250000 0.00000000   1   0.08
> -1

apryor6, 2017-11-08:
Thanks for including the file you are using for input. Could you also tell me the exact call to `prismatic` so I can try to replicate?

matkraj, 2017-11-08:
`prismatic -i "STO.xyz" -t 20 20 40 -o "STOsim.mrc" -wx 0.5 0.5 -wy 0.5 0.5 -a m -4D 1 -3D 0 -F 5 -p 0.05
`

apryor6, 2017-11-08:
The files I get are close to identical, but not identical. There are small floating point differences. As a quick test I amplified the thermal factor from 0.08 to 8.0 and the resulting files are significantly different. Perhaps the original similarity is because the thermal factor is not much larger than 1 pixel, and then you are averaging over 40 unit cells in Z. I could look in more detail later this week.

matkraj, 2017-11-08:
ah ok, was that with a single or a double precision? I did sum(abs(A-B)) and got 0

matkraj, 2017-11-10:
I changed the thermal parameter and FP is clearly working, thanks for the suggestion

closed 2017-11-10 by matkraj