prism-em/prismatic #6: Reducing the number of discs within the 4D output files

pc494, 2017-11-08:
Possibly related to #2 

I'm keen to reduce the number of discs produced within the simulation (rather than just cropping images) - physically something like a meta parameter for "Camera Length" or "Max Scattering Angle".

Can this be done at present?

_Below is an example image and reproducible code to illustrate what I mean._

[image]

[PP_input .TXT](https://github.com/prism-em/prismatic/files/1454429/PP_input.TXT)
MWE example of the code:

```python
import pyprismatic as pr
import numpy as np
from matplotlib import pyplot as plt

meta = pr.Metadata(filenameAtoms="PP_input.XYZ")
meta.save4DOutput = True
meta.save3DOutput = False
meta.scanWindowXMin = 0.495
meta.scanWindowXMax = 0.50
meta.scanWindowYMin = 0.495
meta.scanWindowYMax = 0.50
meta.alphaBeamMax = 0.024/3
meta.probeSemiangle = 0.02/5
meta.tileX=15*3
meta.tileY=15*3
meta.tileZ=105

meta.go()

output = pr.fileio.readMRC('output_X0_Y0_FP1.mrc')
output = output.reshape([output.shape[1],output.shape[2]])
output = np.fft.fftshift(output)
plt.figure()
plt.imshow(np.power(output,0.25),cmap='viridis')
plt.draw()
plt.savefig('image.png')
```

apryor6, 2017-11-08:
This cannot be done at present, but should be straightforward to implement.

cophus, 2017-12-05:
pc494,

One suggestion is that you could try is increasing the "pixel size" in both X and Y.  The maximum outer scattering angle is set by this parameter.  For example, using the default voltage of 80 keV and the default pixel size of 0.1 Angstroms, the max scattering angle is 104.4 millirads.  By changing the pixel size to 0.2 Angstroms, this value will change to 52.2 millirads (half).

Also your simulation cell is thick enough to produce scattering to fairly high index disks.  Your input file looks strange to me:

    2.83   2.83   2.83
31 0.0000 0.0000 0.0000 1.000 0.000
31 0.0000 2.8250 2.8250 1.000 0.000
31 2.8250 0.0000 2.8250 1.000 0.000
31 2.8250 2.8250 0.0000 1.000 0.000
33 2.8250 2.8250 2.8250 1.000 0.000
33 2.8250 0.0000 0.0000 1.000 0.000
33 0.0000 2.8250 0.0000 1.000 0.000
33 0.0000 0.0000 2.8250 1.000 0.000

The 5th row atom is at (2.825, 2.825, 2.825), while the 1st row atom is at (0,0,0).  These atoms are practically on top of each other for cell dimensions of 2.83 A, due to periodicity.  Did you mean for the cell dimensions to be larger?

lerandc, 2021-06-16:
Possible now with 4D Cropping options

closed 2021-06-16 by lerandc