prism-em/prismatic #7: Trouble installing pyprismatic on windows

thomasaarholt, 2017-11-15:
Hi there,

I've been trying to install pyprismatic on two windows machines (one with, one without Nvidia GPU).
I think I've managed to correctly set up the environmental variables, but I'm getting an error to do with the Visual Studio compiler. I've tried doing this with both VS 2017 and 2015 (the error from the latter is in the gist below).

https://gist.github.com/thomasaarholt/c3dd63c549f6f62fb8f715e7dbd8a048

Could you please take a look at the attached gist and see if you have any suggestions?

I'm using the [pre-compiled fftw3 dlls](http://www.fftw.org/install/windows.html), and the normal boost library.

apryor6, 2017-11-15:
The line:

`error C3861: 'time': identifier not found`

indicated to me that there was a header missing. I updated "meta.h" to `#include <time.h>`, which hopefully fixed the issue, though I am not near a Windows machine currently  to verify myself. `PyPrismatic` has also been updated, so try again now. You may need to add `--no-cache-dir` to the `pip install` command to prevent it trying to use the cached version (the newest one is 1.1.16). Also, I'm almost positive VS 2017 produces further errors, so I would continue to stick with 2015 as you mentioned.

thomasaarholt, 2017-11-15:
Brilliant! That did it! I've only attempted the pip install, not compiling prismatic itself.

I'm going to make a little reply below containing what I had to do to make it all work. It actually worked fine on the 2017 install.

apryor6, 2017-11-15:
Awesome, glad it worked.

thomasaarholt, 2017-11-15:
Thanks so much for the quick response. My colleagues were wowed by the response time ;)

thomasaarholt, 2017-11-15:
**The following is an installation guide for pyprismatic on 64-bit windows.**
I've only included the necessary files for the CPU version to install without issue.

Download Visual Studio [2015 ](https://www.visualstudio.com/vs/older-downloads/)or [2017 ](https://www.visualstudio.com/downloads/)- Install the C++ package (one of the first package options). It's about 5GB.
[Download the pre-compiled fftw3 package](http://www.fftw.org/install/windows.html).
[Download boost](https://dl.bintray.com/boostorg/release/1.65.1/source/boost_1_65_1.zip).

Having installed VS we need to add the path to `lib.exe` to our [PATH variable in order to link it](https://www.computerhope.com/issues/ch000549.htm) to VS. 
For VS2017 the path is `C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.11.25503\bin\Hostx64\x64`
For VS2015 the path is `C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\amd64`.

This lets us use lib.exe to create library files out of the fftw3 files. Open a new cmd window, change directory to the fftw folder that was downloaded, and run these three lines:
```
lib /machine:x64 /def:libfftw3-3.def
lib /machine:x64 /def:libfftw3f-3.def
lib /machine:x64 /def:libfftw3l-3.def
```

Create or add the following values to these environment variables. Add each new path separated by a semicolon `;`:
`INCLUDE`: `path\to\fftw-3.3.5-dll64;path\to\boost_1_65_1;`
`LIB`: `path\to\fftw-3.3.5-dll64;`

Now open a new cmd windows and run `pip install pyprismatic`

thomasaarholt, 2017-11-15:
When trying to import pyprismatic I am getting an error:

```python
>>> import pyprismatic
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Miniconda3\envs\pris\lib\site-packages\pyprismatic\__init__.py", line 14, in <module>
    from . import core
ImportError: DLL load failed: The specified module could not be found.
```
This is the content of the folder where it is trying to import `core`.

[image]

I have installed prismatic from the windows binary installer. Should I have built it? I assume pyprismatic acts as a wrapper for the prismatic application?

apryor6, 2017-11-15:
You shouldn't need to install prismatic in order to access pyprismatic (the GUI is just a wrapper around the same underlying code that pyprismatic touches). Are you running python from within the folder containing pyprismatic? Try changing to another directory. This sort of error will appear if you build a python package without `build_ext --inplace` and then try to load it from the same folder.

thomasaarholt, 2017-11-15:
I was already running from another directory. Hm..

thomasaarholt, 2017-11-15:
Looking at similar errors online - it looks like I'm missing a .dll from my path (I thought the core.cp36-win_amd64.pyd file *was* the dll). I may have done this from trying to reduce the files needed to be added to PATH. I'll have a play.

apryor6, 2017-11-15:
What if you add C:\ProgramData\Miniconda3\envs\pris\lib\site-packages\pyprismatic to your PATH?

apryor6, 2017-11-15:
If I recall (and I'm no Windows expert, unfortunately) a .pyd file is very similar to a .dll in Python speak, and it may need to be added to PATH. If that is the case, then I'll need to add this to the docs.

thomasaarholt, 2017-11-15:
That didn't work either. Good try.

What files does pyprismatic access? The entire pyprismatic folder is only 256 KB.

apryor6, 2017-11-15:
For the CPU-only version, the C++ code and Python wrapper are compiled into an extension module (which is what I expect the .pyd to be). Then the python files are the Python interface to that code. The GPU version is similar except that there is also the cuPrismatic.dll which exposes additional functions that access GPU code under the hood.
I'll be able to look more in detail later, but another option would be to download the source code and try installing with `python setup.py build_ext --inplace install`

thomasaarholt, 2017-11-15:
I'll give that a shot. I'm headed home in a bit, but might give this a shot at home with the GPU version.

thomasaarholt, 2017-11-15:
I figured it out! And I feel really stupid, although happy with the way I figured it out. I used [Process Monitor](https://docs.microsoft.com/en-us/sysinternals/downloads/procmon) to find which dlls were being accessed (and filtered for ones that were failing), and realised that I had forgotten to add the fftw3 package to PATH - as you've instructed in your installation procedure.

I've updated my own procedure to account for that mistake.

apryor6, 2017-11-15:
Nice, and thanks for adding that writeup.

closed 2017-11-15 by apryor6

thomasaarholt, 2017-11-16:
**Notes on building fftw on Windows**
Since Windows doesn't have built-in support for a file called `pthreads.h`, we [need to download it](ftp://sourceware.org/pub/pthreads-win32/pthreads-w32-2-9-1-release.zip).
Within the "Pre-built.2" folder, add the dll/x64 (or x86 if 32-bit Windows), include and lib folders to the environment variables PATH, INCLUDE and LIB, respectively.

[Download the latest fftw **source**](http://www.fftw.org/download.html). After unzipping, open cmake-gui and specify the source code as the downloaded folder. Create a "Build" folder in the parent directory. Press Generate. Now select the Visual Studio version you are using AND select the correct architecture - this is probably Win64.

After a little wait, tick on the ENABLE_FLOAT, ENABLE_THREADS and WITH_COMBINED_THREADS options. See screenshot below. The "--enable-shared" option from the Prismatic user guide is called "BUILD_SHARED_LIBS" here, and is already enabled.
Press Generate. Because CMake is a bit weird, we get the following message, but it's not a problem:
```
Looking for pthread.h
Looking for pthread.h - not found
Found Threads: TRUE
```
If you get an actual error here, it may be because you failed to specify the correct architecture when specifying the VS Studio version)

[image]

Press "Open Project". Here, in the menu bar, change the "Debug" text to "Release", then press "Build" -> "Build Solution".

Once completed, you will find the result of the compilation in BUILD/Release. As a final step, copy the "fftw3.h" file from "SOURCE/api/" into BUILD/Release, as this file is needed later.

If one wants to use this for compiling Prismatic, then the options related to fftw are as follows:
[image]

Some notes:
During the compilation in VSCode, I sometimes got a warning that the source files had changed, and had the option to reload. "Ignore All" seemed to work fine.

In order to compile fftw3 in double precision, turn off the "ENABLE_FLOAT" option and compile again.

thomasaarholt, 2017-11-16:
Finally, here are the files I've compiled for 64-bit Windows.
[FFTW3 in single (float), double and long-double precision](https://www.dropbox.com/sh/rf9smo8cnjltnhm/AAAxPsUJN9jPySkVJtdLalbaa?dl=0) with threads and shared library support.
[Prismatic and Cuprismatic compiled with GPU support](https://www.dropbox.com/sh/rf9smo8cnjltnhm/AAAxPsUJN9jPySkVJtdLalbaa?dl=0) but without the GUI (I don't have access to Qt5)