prism-em/prismatic #8: Pyprismatic crashing when running on GPU

thomasaarholt, 2017-11-16:
After having successfully installed pyprismatic with GPU enabled via pip, I am now running into issues when using the GPU for simulation in Python - that is, if I run the Si100.XYZ example with 0 threads, python crashes.

The example file is so small that it only runs on the CPU anyway, so the error isn't immediately noticeable - I noticed it first when I ran pyprismatic on a larger specimen.

[Here is the result in the terminal](https://gist.github.com/thomasaarholt/2b99382b21575d61dd2c9aa4591155dd). I'm not sure its overly helpful. For info, I am running this via Jupyter Notebook.

The larger file runs nicely both using prismatic CLI and GUI.

apryor6, 2017-11-16:
If it runs correctly with the CLI and GUI, that probably means the issue is isolated to Python. What if you run the reconstruction as a regular python script, not through IPython or Jupyter? Could you send a script that reproduces the error?

thomasaarholt, 2017-11-16:
Not working in regular Python from the command line either. Sorry, I meant to check that as I was typing, then forgot...
Here's an example that doesn't work.

[PythonScript.zip](https://github.com/prism-em/prismatic/files/1479100/PythonScript.zip)

apryor6, 2017-11-16:
Confirmed that script crashes (on a laptop with no GPU). It seems the bug is tied to the energy somehow, as if I lower the energy to 200 keV the crash no longer occurs. I'll look into it more

thomasaarholt, 2017-11-16:
Appreciated! As you've probably guessed, our microscope runs at 300kV.

apryor6, 2017-11-16:
Yeah, this is a bug. There’s no reason the code shouldn’t work for any energy

thomasaarholt, 2017-11-16:
Confirmed the behaviour here. 

As a separate issue (I might open a new issue for this):
I'm noticing that I'm running out of PC ram when running the sample file I provided at 80kV though. I get the following error: `GPUassert: out of memory C:/Users/Me/Downloads/Prismatic/prismatic-master/src/PRISM02_calcSMatrix.cu 163`

Not sure why it soaks up so much ram. (I have 16 GB) With the GUI and same settings it uses about 1.5 GB.

apryor6, 2017-11-16:
Just to clarify, you can run the simulation as expected using GPU resources with `prismatic` or `prismatic-gui`, but when you run the same parameters with `pyprismatic` it crashes with out of memory errors?

thomasaarholt, 2017-11-16:
That is correct.

apryor6, 2017-12-01:
Just getting around to sorting this out and preparing for the next release, sorry. I ran the same simulation with `prismatic` and get the same crash, but the reason isn't a bug. It's because the simulation is quite large and the S-matrix occupies a lot of RAM (roughly estimate to be 1500 x 1500 x 7472, which is > 130GB for a complex float array. The simulation you sent is using PRISM, not multislice, so I assume that is the algorithm you want. In this case, you should increase the PRISM interpolation factor. Because this simulation cell is so large, you can afford to crop the probe significantly, which means that you can increase the PRISM interpolation factor by quite a bit and still produce an accurate result.

The `prismatic-gui` Probe Analyzer tab is useful for this purpose so that you can compare the results of a single PRISM probe calculation vs Multislice for different interpolation factor (f) settings and identify a setting that is a suitable tradeoff between speed and accuracy. As a starting point, trying f=16 or f=32 would reduce the array from 130GB to 8GB or 2GB, respectively.

closed 2017-12-01 by apryor6

thomasaarholt, 2017-12-01:
That's great news! I'll have a go at it on Monday! (off for a weekend trip to London!)

thomasaarholt, 2017-12-03:
You're entirely right, and the terminal does in fact report the error.
Any chance we can have the error print the amount of ram it was trying to allocate?
Currently the error message is this:
`GPUassert: out of memory C:/Users/Me/Downloads/Prismatic/prismatic-master/src/PRISM02_calcSMatrix.cu 74`
Had to crank things up to f=128 on my home machine ( gpu with 4 GB ram), in order for it to not saturate the ram.