prism-em/prismatic #9: Load settings in gui

thomasaarholt, 2017-11-17:
(I'm just listing these issues as I'm finding them, things I mention are not necessarily a priority)

It would be nice to be able to save/load the settings used for the GUI, as it can be a bit tedious entering all the values manually each time one wants to run a simulation (especially if one kills the simulation because it ate up all my ram 😅  )

apryor6, 2017-12-01:
I'm a fan of this idea, and am going to try to implement this today. The plan would be to allow for a params.txt file that contains simulation parameters. This will be written out every time a simulation is run from the gui, and the last params file will be loaded at startup. I'll also add a button to load a custom file.

This would be particularly nice because users could share an .xyz file and a params file with a colleague and you can be confident that you are replicating the exact same simulation.

apryor6, 2017-12-01:
Didn't mean to close -- I'll close in the relevant commit once this is implemented.

commit 9b48bee3 (2017-12-01): Closes #9 -> Added functionality for parameter file reading/writing
commit 0af04eee (2017-12-01): Closes #9 -> Added functionality for parameter file reading/writing
closed 2017-12-01 by apryor6