prism-em/prismatic #11: New release in the next week(s)?

thomasaarholt, 2017-11-28:
I'm about to request that Prismatic be added to our university's (Oslo, Norway) program kiosk for installation university-wide. (don't expect _thousands_ of new users though ;) ). Before I do that, I thought I'd ask if you were planning a new release soon, as it would save some work not having to update it again soon.

apryor6, 2017-11-28:
That's great to hear. I do plan to have a new release soon, but it is only a minor version and is mostly small changes / bug fixes for the GUI. I also wanted to sort out some of the existing open issues. I've been a bit busy and lately my Windows machine has been causing trouble so I'll need to sort that out to rebuild installation packages to release. It should hopefully happen soon though.

apryor6, 2017-12-01:
I've finished the latest update, so if you are compiling from source you should be able to go ahead and install. The Mac OSX installers are [here](http://prism-em.com/installers/), and the Windows ones will be up with the next couple days when I get around to rebuilding them.

closed 2017-12-01 by apryor6