prism-em/prismatic #12: Probe analyzer does not compute difference image in real space correctly

cophus, 2017-12-05:
Difference image in Fourier space looks correct, but in real space there must be some normalization error.  See this image using tutorial xyz file as base:

[image]

apryor6, 2017-12-05:
The probe images are normalized by the total sum, see below:

https://github.com/prism-em/prismatic/blob/78b2438b92d6322ed3000818baf86b1fe2ca104d/Qt/prism_qthreads.cpp#L217-L227

And then the contrast for the difference image is set based upon the min/max values in the multislice probe

https://github.com/prism-em/prismatic/blob/78b2438b92d6322ed3000818baf86b1fe2ca104d/Qt/prismmainwindow.cpp#L1511-L1527

Where the pixel color is determined by linearly interpolating from the colormap

https://github.com/prism-em/prismatic/blob/78b2438b92d6322ed3000818baf86b1fe2ca104d/Qt/prism_colormapper.cpp#L10

It does look like something is amiss in the image you attached, however. Hmm

alex-rakowski, 2021-06-16:
check QT-window