prism-em/prismatic #15: Strange results for stacked cells

thomasaarholt, 2017-12-09:
Hi,

I've been trying to use Prismatic to replicate simulation results that I have previously done with [MULTEM](https://github.com/Ivanlh20/MULTEM).
I export my specimen file via ASE in python, and have written a writer to take a given [ASE](https://wiki.fysik.dtu.dk/ase/index.html) atoms object and write a prismatic specimen file. I simultaneously spit out a larger supercell .cif file to check with [Vesta](http://jp-minerals.org/vesta/en/) that things look correct.  

When I simulate a reasonably basic model of 100 ZnO at 300kV, I find that at medium acceptance angles (25 - 50), the image looks how I would expect:

[image]

But, at higher acceptance angles (65-99), things start looking very strange:
[image]

This is unlike what I find with MULTEM (can't attach an image atm). I've experienced this result using both the PRISM and Multislice algorithms in prismatic.

I'm attaching the specimen file and the parameter file (nice to have that :) ) for reproducibility. Included is also a cif file exported for viewing in software like [Vesta](http://jp-minerals.org/vesta/en/).
[Paper1.zip](https://github.com/prism-em/prismatic/files/1544808/Paper1.zip)

Could someone comment on this? It could very well be me doing something wrong, but I've spent all day trying to locate my mistake.

apryor6, 2017-12-11:
Although you said that you experienced this issue with both PRISM and Multislice in `Prismatic`, I suspect this issue is limited to the PRISM algorithm. If you can reproduce with Multislice, however, feel free to send along the simulation configuration.

These artifacts are because your PRISM interpolation factor is too high. A high interpolation factor effectively multiplies the probe by a binary square mask and introduces a aliasing sinc function in reciprocal space. This artifact becomes greatly exaggerated at higher scattering angles. If I use your parameter file but change the interpolation factor to f=4 the image looks okay

[image]

The probe module is good for visualizing the effect of the current value of the interpolation factor. With f=16 the probe produced by PRISM is extremely incorrect and you can see the sinc-aliasing I mentioned previously. 

[image]

Whereas with f=4 the result is much closer and the various quantitative metrics reflect that as well.

[image]

Keep in mind the interpolation factor determines how tightly the probe is cropped in real space. As long as the crop is chosen such that most of the probe intensity is included, the images it produces will be accurate. These relevant papers go into much greater detail:

[Prismatic code](https://link.springer.com/article/10.1186/s40679-017-0046-1)
[PRISM algorithm](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5656717/)

apryor6, 2017-12-11:
@cophus -- this is related to our discussion of frequency-dependent artifacts for low interpolation settings a few weeks (months?) ago.

closed 2017-12-11 by apryor6

thomasaarholt, 2017-12-11:
Thanks for the great explanation! Looking at the multislice result I ran over the weekend, you are completely correct. :)