prism-em/prismatic #18: Linux?

thomasaarholt, 2017-12-19:
I'm looking at using the [Norwegian Abel supercomputer](http://www.uio.no/english/services/it/research/hpc/abel/) for prismatic, mainly in order to have enough ram for large computations. They, however, run Linux. 

Has there been any attempt at compiling prismatic for linux?

apryor6, 2017-12-19:
Absolutely. My primary OS is Ubuntu 16.04, and I have also compiled it on clusters running red hat and scientific linux. However, the assumption is that if you are running such a system you (or a sysadmin) are capable of compiling from source.

thomasaarholt, 2017-12-19:
Ok, excellent. I'm no expert, but I've had some experience with (the difficulties of) compilation in the past few months. Could you take a quick look at the "[available software](http://www.uio.no/english/services/it/research/hpc/abel/help/software/#toc2)" list and see if you recognise any of the compilers?

thomasaarholt, 2017-12-19:
Btw, the nodes we are using run the Linux 64 bit CentOS 6. I'm totally unfamiliar with it, but we'll see.

apryor6, 2017-12-19:
The language list must incomplete, as supporting CUDA necessitates that you support C++ or at least C. CUDA requires the gcc compiler (as detailed [here](http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html#verify-that-gcc-is-installed)), which in my experience is the most commonly available/used C/C++ compiler on Linux. I would be very surprised if your cluster supports CUDA but does so without the use of gcc.

Assuming gcc is available, but the Intel compilers are the default, you may find you have to do a little more work than if gcc is the default. I ran into this issue before. For example, the cluster may have preinstalled many libraries such as FFTW for you, but if they are compiled with the Intel compilers then they will not work. In my case I just had to make sure that the Intel compilers were unloaded, the gcc ones were loaded, and then I compiled `FFTW` + `prismatic` using that setup and everything worked fine.

I appreciate the fact that these types of processes can seem absurdly convoluted and daunting, but usually they are simpler than they appear. I'm happy to help if you run into specific compilation errors.

thomasaarholt, 2017-12-19:
That does make me feel a bit better, thank you :) And thanks for the comments!

thomasaarholt, 2017-12-20:
You're entirely right. I just got access to the cluster, and gcc is here.

thomasaarholt, 2017-12-20:
I've compiled fftw successfully using the gcc compiler. I wrote it to a directory in my home directory as I didn't dare mess things up with `sudo` quite yet.

When running `cmake .` in the prismatic-master directory I am getting 

```
FFTW_EXECUTABLE_LIBRARIES = fftw3ffftw3f_threads                                            
FFTW_LIBRARY = fftw3ffftw3f_threads                                                         
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:          
FFTW_INCLUDE_DIR                                                                            
   used as include directory in directory /usit/abel/u1/thomasaar/prismatic-master          
```

I attempted setting `FFTW_INCLUDE_DIR` by `export FFTW_INCLUDE_DIR='$HOME/build_float/include'`, without success, so I'm assuming it is set in the cmake files. I'm noticing the file FINDFFTW3.cmake in the `cmake/modules` directory, but it looks like it's looking for the FFTW_INCLUDE_DIR environment variable (which I just set), so I'm a little lost on where to proceed to.

apryor6, 2017-12-20:
There's many ways to solve this, but you may find [this section](http://prism-em.com/installation/#from-source) of the docs helpful.

I'm almost positive you cannot set CMake variables using `export` as you tried to do. The way to set `FFTW_INCLUDE_DIR` is to invoke `ccmake .` within the build directory and adjust it or to add `-DFFTW_INCLUDE_DIR='$HOME/build_float/include` to the `cmake` command, depending how you are invoking.

thomasaarholt, 2017-12-20:
Ah yes. I now remember doing this via the cmake gui on windows. Thanks for your help, Alan :)

thomasaarholt, 2017-12-20:
Apologies for not looking properly at the documentation earlier. I've been spending a bit of time trying to get the following to work:

```
-bash-4.1$ cmake ../
-- Boost version: 1.41.0
-- Found FFTW threads: $HOME/build_float/lib
-- Found FFTW: $HOME/build_float/lib
Boost dir /usr/include
GPU support enabled
CMake Error at /cluster/software/VERSIONS/cmake-3.7.1/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find CUDA (missing: CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (found
  version "8.0")
Call Stack (most recent call first):
  /cluster/software/VERSIONS/cmake-3.7.1/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /cluster/software/VERSIONS/cmake-3.7.1/share/cmake-3.7/Modules/FindCUDA.cmake:1013 (find_package_handle_standard_args)
  CMakeLists.txt:79 (find_package)
```
Is this something that you've run into before?

For now, I've got the following cmake parameters:
[image]

I can't tell if the CUDA SDK and Toolkit should not refer to the same folder. For now they both refer to the root of the cuda install (v8) we have here. I suspect that the sdk parameters is outdated and not needed?

thomasaarholt, 2017-12-20:
Scratch that, we're good:

`cmake -DCUDA_CUDART_LIBRARY=CUDA_CUDART_LIBRARY -DCUDA_TOOLKIT_ROOT_DIR=$CUDA /cluster/home/thomasaar/prismatic-master`

apryor6, 2017-12-20:
Great. The other confusing thing is that sometimes cmake variables are listed not found but are also not necessary (like the mpi library here), and other times specifying just one of the variables will allow cmake to auto locate other similar ones.

thomasaarholt, 2017-12-20:
I'm a bit concerned that the version of gcc we are using is old. Could you check `which gcc` on yours? Ours is 5.1.0.

thomasaarholt, 2017-12-20:
Okay, things are looking up. I ran into an issue of fftw3.h not being found, despite being pointed to correctly by the `FFTW_INCLUDE_DIR` cmake parameter. I solved that by simply copying it to `pristmatic/include`.

However, when running cmake I am missing `-lfftw3f`. Any suggestions?

```
[ 78%] Building CXX object CMakeFiles/prismatic.dir/src/utility.cpp.o
[ 78%] Building CXX object CMakeFiles/prismatic.dir/src/projectedPotential.cpp.o
[ 84%] Building CXX object CMakeFiles/prismatic.dir/src/parseInput.cpp.o
[ 94%] Building CXX object CMakeFiles/prismatic.dir/src/go.cpp.o
[ 94%] Building CXX object CMakeFiles/prismatic.dir/src/atom.cpp.o
[100%] Linking CXX executable prismatic
/usr/bin/ld: cannot find -lfftw3f
collect2: error: ld returned 1 exit status
make[2]: *** [prismatic] Error 1
make[1]: *** [CMakeFiles/prismatic.dir/all] Error 2
make: *** [all] Error 2
``

thomasaarholt, 2017-12-20:
Right, I'm certain it's because I didn't `sudo make install` and instead `make install` into a specific directory (which was not usr/local/bin or similar). I might give `sudo` a shot, since I don't know how to link -lfftw3f to whatever file it's looking for after building.

apryor6, 2017-12-20:
Add the directory containing the fftw3f libraries to `LIBRARY_PATH` (these are the files named something like libfftw3f.so

Since you compiled the libraries locally this path is not on the default search, hence the error

apryor6, 2017-12-20:
You'll also need to add the paths to `LD_ LIBRARY_PATH` so they can be found at runtime. FYI, `LIBRARY_PATH` is for compile time, `LD_ LIBRARY_PATH` is for runtime. I would recommend adding the associated `export` calls to the end of your ~/.bashrc file so they are persistent

thomasaarholt, 2017-12-20:
Aye, I'll be adding them to `~/.bashrc`.

It was a bit optimistic to try to manage all of this today - I'm flying out to the UK for ten days of xmas tonight. I may be remoting in to have a go during that time, however.

Thanks for your help, Alan, and merry christmas!

apryor6, 2017-12-20:
As promised, compiling can be quite the headache.

No problem, enjoy the holidays

thomasaarholt, 2017-12-24:
Ssh over ipad seems to working decently. Have you encountered this error in your travels? It's when running 'make'. 
[image]

thomasaarholt, 2017-12-24:
Right, afaik, the error was partly a result of an old version of ld (the GNU linker). I have access to a newer version of binutils, and running with the version of ld in that (2.28 rather than 2.20), results in less of an error, but now with essentially no error message. Any tips? 

I am not running this as sudo.

```
CMakeFiles/prismatic.dir/src/atom.cpp.o: In function `std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >::~basic_stringbuf()':
atom.cpp:(.text._ZNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEED0Ev[_ZNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEED5Ev]+0x3): undefined reference to `vtable for std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >'
CMakeFiles/prismatic.dir/src/go.cpp.o: In function `Prismatic::go(Prismatic::Metadata<float>)':
go.cpp:(.text+0x365): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned long)'
go.cpp:(.text+0x3f9): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long)'
collect2: error: ld returned 1 exit status
make[2]: *** [prismatic] Error 1
make[1]: *** [CMakeFiles/prismatic.dir/all] Error 2
make: *** [all] Error 2
```

apryor6, 2017-12-24:
Whenever changing compilers I think it’s a good idea to delete and remake the build folder to prevent any issues with a cached setting from the old compiler being remembered. The linking error you’re getting looks like it’s confused about a c++11 feature, which could be caused by an outdated compiler issue like you are describing.

closed 2018-08-14 by apryor6

thomasaarholt, 2018-11-15:
I had another go at this on the supercomputer today (just been running on a single GPU computer so far), and made leaps and bounds of progress in minutes. I have at the very least installed the CPU version, and I think the error I get on the GPU version is because the GPU isn't made available yet.

Our system defaults to intel compilers. unloading the intel compilers and loading gcc 4.9.2 was necessary. I also had to specify a "manual" path to gcc's c and c++ compiler to the cmake command, i.e.
`cmake ../ -DCMAKE_C_COMPILER=/cluster/software/VERSIONS/gcc-4.9.2/bin/gcc -DCMAKE_CXX_COMPILER=/cluster/software/VERSIONS/gcc-4.9.2/bin/c++
`
I also had to make sure that other compilers were not selected in the CCMake gui settings.

Hopefully the above can be of use to someone else one day.

apryor6, 2018-11-15:
I can echo this. I previous compiled on a system that defaulted to Intel compilers and had to take similar steps. It seems `nvcc` does not play nicely with these [(more here)](https://devtalk.nvidia.com/default/topic/531581/using-intel-c-c-compilers-with-cuda-tools/).

Some clusters have convenience utilities in place to swap compilers, such as `module unload icc` `module load gcc`. If not, you will need to manually modify the CMake options as @thomasaarholt  mentioned above. You might also need to modify the `PATH` to ensure the right version of `gcc` is found first.

thomasaarholt, 2018-11-15:
I ended up with 
`module purge`
`module load gcc/4.9.2`
`module load boost`
`module load cmake`

Currently I'm trying to link pyprismatic/cuprismatic to my built GPU-prismatic.

After installing using setup.py (successfully), I'm now getting
```
>>> import pyprismatic as pr
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/cluster/home/thomasaar/prismatic/pyprismatic/__init__.py", line 14, in <module>
    from . import core
ImportError: cannot import name 'core' from 'pyprismatic' (/cluster/home/thomasaar/prismatic/pyprismatic/__init__.py)
```

I encountered this in #7, but "fixed" it by adding the fftw directories to my PATH.

How does pyprismatic know where my cuprismatic files are?

apryor6, 2018-11-15:
Change to a different directory (anything other than the source) and try the import again.

thomasaarholt, 2018-11-15:
Yes!

apryor6, 2018-11-15:
This happens because although the install was successful you were in a directory with a folder of the same name as the package you wanted to import, which messes with the order of operations used for importing. The solution is to change directories or to modify ` pythong setup.py install <...other options>` to   setup.py build_ext --inplace install <...other options>`

thomasaarholt, 2018-11-15:
Right, any idea why I'm getting 
```python
>>> pr.demo()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usit/abel/u1/thomasaar/.conda/envs/py/lib/python3.7/site-packages/pyprismatic/__init__.py", line 35, in demo
    meta.go()
  File "/usit/abel/u1/thomasaar/.conda/envs/py/lib/python3.7/site-packages/pyprismatic/params.py", line 270, in go
    pyprismatic.core.go(*(l))
TypeError: must be real number, not str
```
I think it's reacting to the following parameter of metadata:
`transferMode : memory model to use, either “streaming”, “singlexfer”, or “auto”`

apryor6, 2018-11-15:
I'll have to check, but it is most likely an issue with parsing one of the default arguments and/or a difference in the GPU vs CPU code that contains some bug. What version of Python? Does the error occur on the CPU only version?

thomasaarholt, 2018-11-15:
I just tried downgrading to python 3.6 from 3.7. 3.6 is working on my single-GPU computer.

I do get the message "COMPILED FOR GPU", implying that the error comes about in the cpp code, [after this point](https://github.com/prism-em/prismatic/blob/a989cd0d08bcb282a019e88eb9717b935e8a31df/pyprismatic/core.cpp#L42).

thomasaarholt, 2018-11-15:
Found the bug and fixed it, in #42

thomasaarholt, 2018-11-15:
A final question from me - have you met this before?
```python
#after calling pr.demo()
save3DOutput = true
save4DOutput = false
numGPUs = 4
numStreamsPerGPU = 3
alsoDoCPUWork = 1
earlyCPUStopCount = 100
Data Transfer Mode : Auto
Execution plan: Multislice
Using GPU codes
Estimated potential array size = 556032
Estimated buffer memory needed = 3686400
meta.numStreamsPerGPU*2*batch_size*imageSize[0]*imageSize[1]= 307200
GPUassert: CUDA driver version is insufficient for CUDA runtime version /usit/abel/u1/thomasaar/prismatic/src/configure.cpp 157
```

I'm hoping the GPUerror is because my user on the supercomputer does not currently have access to a GPU, but I don't really think that's it. Ideas?

thomasaarholt, 2018-11-16:
I suspect that it's a problem with me [using ssh from a computer without CUDA](https://devtalk.nvidia.com/default/topic/977480/jetson-tx1/cuda-driver-version-is-insufficient-on-ubuntu-16-04/post/5021493/#5021493), but it's hard to be sure...

thomasaarholt, 2018-11-16:
Perfect, thanks. I'll tell the guys to fire up the supercomputer :p

>

apryor6, 2019-01-08:
Welcome!

Your GPU is new enough. According to https://developer.nvidia.com/cuda-gpus the GTX 760 is a compute capability 3.0, which meets the requirements of Prismatic http://prism-em.com/installation/

The error you are receiving either means you do not have an updated driver of the correct version installed or the GPU is unreachable for some other reason. The `nvidia-smi` command is installed when you install the CUDA toolkit and is a nice tool to verify your computer is able to communicate with the GPU. I would run that command first to verify things are set up as you expect.

phyxie, 2019-01-08:
Thanks for the quick response, and some really neat software!
I was unaware that I needed the CUDA toolkit to run the GPU version, since installing it's all working perfectly.
I'm getting a copy of crystalmaker to generate some cool structures with, so hopefully I'll see some pretty  simulations soon.

yobc401, 2019-06-08:
Hello, I'm a newcomer for living Prismatic world. So far I'm trying to install the program in my linux system (Ubuntu 18.04 LTS). I followed the instruction on the web site, and I succeed to set the environment and invoke Cmake for installing GUI option.
The result is the following:
 -- Boost version: 1.65.1
-- Found FFTW threads: /usr/lib/x86_64-linux-gnu/libfftw3_threads.so
-- Found FFTW: /usr/local/lib/libfftw3.a
Boost dir /usr/include
Qt5Widgets_DIR=/home/yeo/anaconda3/lib/cmake/Qt5Widgets
Single precision enabled
UNIX
FFTW_EXECUTABLE_LIBRARIES = fftw3ffftw3f_threads
FFTW_LIBRARY = fftw3ffftw3f_threads
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yeo/Downloads/prismatic-master/build

But, when I command "make -j 8", it would become failed with lots of error messages. The one example is like this.
/home/yeo/Downloads/prismatic-master/Qt/prism_qthreads.cpp:389:79: error: no matching function for call to ‘Prismatic::ArrayND<3, std::vector<float> >::ArrayND(Prismatic::Array4D<float>&)’
         Prismatic::Array3D<PRISMATIC_FLOAT_PRECISION> net_output(params.output);
                                                                               ^
In file included from /home/yeo/Downloads/prismatic-master/Qt/prismmainwindow.h:28:0,
                 from /home/yeo/Downloads/prismatic-master/Qt/prism_progressbar.h:21,
                 from /home/yeo/Downloads/prismatic-master/Qt/prism_qthreads.h:17,
                 from /home/yeo/Downloads/prismatic-master/Qt/prism_qthreads.cpp:14:
/home/yeo/Downloads/prismatic-master/include/ArrayND.h:35:13: note: candidate: Prismatic::ArrayND<N, T>::ArrayND() [with long unsigned int N = 3; T = std::vector<float>]
             ArrayND(){};
             ^~~~~~~

At first, I suspect the mismatching of the Qt version. But, the Qt version in my case is also 5.9.0, but it cannot be proceed. Please, could you help me this problem?

thomasaarholt, 2019-06-08:
@yobc401 could you paste the full error message into a public gist (gist.github.com) and link it here?

thomasaarholt, 2019-06-08:
I have not tried building prismatic since #54 was merged. It *could* have introduced your error, although the error shouldn't lie below /Qt/, since that hasn't been changed in a year.

If you would like to try, build the previous version of prismatic from here: https://github.com/prism-em/prismatic/archive/14aa0244fd1bef6223a803b954d0882f828b60da.zip

yobc401, 2019-06-09:
Thank you for replying to me. I linked it following: https://gist.github.com/yobc401/6110ae2b9279f1510948d33f6df6fa2f.

I tried to install the code( https://github.com/prism-em/prismatic/archive/14aa0244fd1bef6223a803b954d0882f828b60da.zip) that you linked above. When I commanded "make -j 8", it is proceeded. But when I run "prismatic", the program is failed as following: https://gist.github.com/yobc401/2137494d5befc3e4194ca8ccec27bbe4

Once I'd like to succeed to run GUI version of prismatic, then I would be able to test prismatic.

What do you think about my errors?

yobc401, 2019-06-10:
I tried to invoke cmake in the /Qt/ directory, and I succeed to install it. Thanks a lot.

bryandesser, 2019-06-25:
@thomasaarholt  @yobc401 I can successfully build prismatic with any/all options except  the GUI. I am seeing the exact same errors @yobc401 [posted](https://gist.github.com/yobc401/6110ae2b9279f1510948d33f6df6fa2f) a couple weeks back. @yobc401 what do you mean about invoking `cmake` in the `/Qt/` directory? I've tried invoking `cmake` from both `/build/` and `/Qt/` and both come out with the same errors (`cmake` runs fine, `make` starts throwing errors at `prism_qthreads.cpp`). 

Using:
`cmake -DPRISMATIC_ENABLE_CLI=1 -DPRISMATIC_ENABLE_GPU=1 -DPRISMATIC_ENABLE_GUI=1 -DPRISMATIC_ENABLE_PYTHON_GPU=1 ../; make -j 8`

I'm running Ubuntu 18.04, all dependencies seem to be correctly set up as the CPU and GPU command line interfaces run just fine. Qt is version 5.13.0