prism-em/prismatic #19: Prismatic crashes after force-quitting + folder typo bug

thomasaarholt, 2018-01-04:
I had to stop a simulation that was hitting my ram ceiling (by ctrl alt del). Opening prismatic now yields the following error:
[image]

The text file is the following. Note that there is a folder name bug here (I'm guessing its a result of using os.path.join with a plus instead of a comma), as Roaming should be the parent directory, not the first part of the filename.

[Roamingprismatic_gui_params.txt](https://github.com/prism-em/prismatic/files/1603366/Roamingprismatic_gui_params.txt)

As to the crashing behaviour I can only guess. I noticed that C3 and C5 only have one dash - in front of them (all other parameters have two --), although changing this does not affect the crashing.

Removing the file does correct the issue.

cophus, 2018-07-01:
I had this issue happen yesterday - and removing the file also corrected it for me.  This happened during a teaching session!

thomasaarholt, 2018-07-01:
Well, I'm glad others find my bugs too!
Say hello to Monika (Norwegian PhD student) at SuperSTEM for me!

closed 2021-06-16 by lerandc