prism-em/prismatic #22: pyprismatic metadata alphaBeamMax default value

thomasaarholt, 2018-02-21:
The default value for alphaBeamMax is 0.024, though the docs say it should be in mrad.
I'm not sure how that has been impacting my previous simulations.

```python
>>> import pyprismatic as pr
>>> pr.Metadata().alphaBeamMax
0.024
```

thomasaarholt, 2018-02-21:
The same goes for `integrationAngleMin` and `integrationAngleMax`, although I haven't been using those.

apryor6, 2018-02-21:
Typo in the docs as there is no implicit conversion done like there is in the prismatic cli. There is some internal sanity checking performed, so for the case where you entered probeSemiangle = 20 (rad), it would have been truncated to whatever the maximum actual value was, so that is why you still get some result.

closed 2018-02-21 by apryor6

thomasaarholt, 2018-02-21:
Ah! Thanks, I was about to comment on your second point after noticing your commits. Well done anticipating :)