prism-em/prismatic #24: FP > 1 without Thermal Effects

thomasaarholt, 2018-02-26:
In prismatic and pyprismatic, if I turn thermal effects / temperature off, but have more than one frozen phonons, the software will repeat the measurement for each new frozen phonon mode, even though the measurement will be identical.

That is, I believe it makes no sense that with Thermal Effects off, Prismatic iterates over more than one frozen phonon.

Or am I missing something?

apryor6, 2018-02-27:
This is true but is also what one would expect. If thermal effects are turned off, there is nothing to be gained by performing multiple frozen phonons configurations. Perhaps what you are suggesting is we should add a check and display some warning message if the user requests multiple frozen phonons with temperature effects off?

thomasaarholt, 2018-02-27:
Yes, essentially. Actually, that might make more sense than what I was thinking. I was suggesting that prismatic would run only one "phonon" (static) configuration.

However, as you may be insinuating, that might trick the user to think that it has performed with phonons on. So yes, a warning would be apt :)