prism-em/prismatic #26: all-zero image in 3D output

ialxn, 2018-03-09:
Dear all

I seem to get an all-zero image in my output every time at index ``1``

```python
In [1]: data=pr.fileio.readMRC('output.mrc')

In [2]: data.shape
Out[2]: (330, 591, 208)

In [3: np.all(data[:, :, 1] == 0)
Out[3]: True
```
For all other indices in the last place I obtain an image

```python
In [4]: np.all(data[:, :, -1] == 0)
Out[4]: False
```
So, is image ``1`` (and maybe ``0``) special?

4.4.0-116-generic x86_64 GNU/Linux
gcc version 5.4.0 20160609
cuda 9-1

thomasaarholt, 2018-03-09:
I do not get the above. I have normal, nonzero (but small) values.
You should expand a bit on your question: 
What are you simulating? What are your parameters? What does your image look like?

Post the output of:
```
meta = pr.Metadata(your_values))
meta.toString()
```
And maybe upload your model?

apryor6, 2018-03-09:
This is a simulation-specific phenomenon. The dimension you are referring to is that of the radial binning for integration and is affected by the choice of `detectorAngleStep`. Depending upon the choice of this parameter and the resolution scale of the simulation, there can be empty bins (i.e. the detector is pixelated but the number of radial detectors may be chosen to be arbitrarily small). Generally, final images will be produced from these simulations by further integrating between an inner and outer angle, such as to produce a HAADF image, at which point this effect becomes entirely unimportant. 

The following two snippets of code are relevant to this topic, if you are interested.

https://github.com/prism-em/prismatic/blob/4361422b410fff6b921381c88f1947528bc7575a/src/PRISM03_calcOutput.cpp#L75-L83

https://github.com/prism-em/prismatic/blob/4361422b410fff6b921381c88f1947528bc7575a/src/PRISM03_calcOutput.cpp#L369-L373

ialxn, 2018-03-12:
OK, this makes sense. Thanks for the clarifications.

closed 2018-03-12 by ialxn