prism-em/prismatic #27: quastion about the sign of defocus

matkraj, 2018-03-15:
Hi, what is the sign of the defocus? I guess 0 is top of the sample. Is the bottom df<0 or  df>0?
Thanks

cophus, 2018-03-16:
Hi,

We use the same sign convention for defocus C1 as we do for the propagator / spherical aberration terms, i.e. the equation:

chi(q) =  - i * pi * lambda^1 * q^2 * C1 ....
      - 0.5 * i * pi * lambda^3 * q^4 * C3 ....
  - 0.333 * i * pi * lambda^5 * q^6 * C5 ....

Which is the same sign as the propagator function:

chi(q) =   - i * pi * lambda^1 * q^2 * dz.

So changing "defocus" moves the probe in the same direction as the z axis - And you are correct that "0" defocus is the top of the sample.  Setting it to the same value as the z height of the cell moves it to the bottom of the sample.  And setting it to a negative value moves the probe to a focal plane above the sample entrance surface.

Since most authors use the convention df = -C1, we should probably be clearer about our labeling and the direction convention.  Thanks for pointing this out!

matkraj, 2018-07-20:
Hi,
Very late thanks for the information. I will close this.

closed 2018-07-20 by matkraj