prism-em/prismatic #29: Is alphamax calculated correctly?

thomasaarholt, 2018-04-04:
Short:
Is there a factor of 2 missing in the following expression? (and in these two files: [multislice](https://github.com/prism-em/prismatic/blob/3278bdddd34e84f8f70966190b3b98be70c86856/src/Multislice_calcOutput.cpp), [prism](https://github.com/prism-em/prismatic/blob/3278bdddd34e84f8f70966190b3b98be70c86856/src/PRISM03_calcOutput.cpp))
https://github.com/prism-em/prismatic/blob/4ea9bf80155e5051ea0532a98de7a93c632ec677/Qt/prismmainwindow.cpp#L1774

Long:
I've been reading Kirkland: Advanced Computing in EM, and under `5.5.2 Thin Specimen Images` (page 108 in the latest edition) he refers to:
`Nx > 2*a*theta/lambda`
In this example, he's calculating the number of pixels that the potential must be divided into for a `theta=200mrad`, `lambda=2.5pm` and `a=26.89Å`. `theta` is the maximum scattering angle and `a` is the length of the specimen in the x-direction.

He then calculates that to be 429 pixels. That's equivalent to a pixel size of `26.89Å/429 = 0.063Å`.

But when I enter these values into Prismatic, the alphamax is off by a factor of two:
[image]

Am I doing something wrong?

cophus, 2018-04-09:
Hi Thomas,

Assuming I understand your question, no it is correct - but we need to better explain what is being reported.  **Both** Kirkland's expression and our reported numbers are correct.  Kirkland is showing the maximum possible scattering angle of the simulation, which is defined only by lambda (which sets inverse pixel size) and the simulation pixel size (0.063 A in this case). This is just the outermost edge of reciprocal space, since if we move one pixel further from +200 mrads, we land on -200 mrads (periodicity of the Fourier transform).

However in our simulation we use an "anti-aliasing aperture" at 0.5 times the value of qMax / alphaMax.  This is simply a mask where electron intensity is set to zero at every propagation step. So while the simulation can contain electrons at these angles (briefly), they will never be output.  Instead we are reporting the maximum non-zero output scattering angle of the simulation, which is 0.5 times your expected value.  Hopefully this answers your question.  AJ - we could changes the GUI text to be: "output a max" maybe?  Or I could just start another markdown page and actually document all of the labels and inputs (this is probably the better idea haha).

Kirkland later discusses his (2/3,2/3) anti-aliasing aperture - I think this is a bad idea for several reasons, and so I use a (1/2,0) anti-aliasing aperture in all my codes. Thomas if you would like more details I would be happy to post more info.

thomasaarholt, 2018-05-03:
Hi Colin!
Sorry for the long wait! And thank you so much for your patient and long reply - it explains perfectly my issue with being off by a factor of two, and furthers my understanding of prismatic.

closed 2018-05-07 by thomasaarholt