prism-em/prismatic #31: Do thermal effects offset the atoms also in the z-position?

thomasaarholt, 2018-06-07:
Hello again,

When including the effect of temperature through the frozen phonon configurations, the atomic lattice is randomly offset in some direction according to the atoms' Debye-Waller factor.

Does Prismatic apply this offset only in the x-y plane, or also in the beam direction (z)?

The reason I ask is that I am wondering about the impact this should have on the slice thickness. Working with a unit cell or super-cell with a given c-length, it is tempting to set the slice thickness to be some fraction of c, for instance c/6, should there be six obvious "layers" to the cell. (In my case, this distance is equivalent to about 1.625Å)

However, if the thermal effect shifts the atom along the beam direction, it may be "swallowed" entirely into the slice above or below it. If the thermal effect is only applied in the xy plane, then this possible problem is avoided, but the simulation is less "physical".

Prismatic with the PRISM algorithm is extremely fast, and I can happily simulate with a much smaller slice thickness at the cost of time, but I'd rather spend that time on a higher number of frozen phonons (200).

What are your thoughts on this? 
Thanks for taking the time to respond to my questions!

Best,
Thomas

cophus, 2018-06-08:
Hi Thomas,

Nope, we have not implemented any z position RMS atomic shifts.  I have however implemented it in my 3D multislice code, but that is unfortunately matlab-only currently.  Let me make 3 comments on your comments:

1 -  Overall I would say you are exactly correct.  We specifically did not use z-plane shifts because it would be unphysical for atoms to make a 1.6 A "jump" into the plane below, especially if they merged with an atom already there!  Or if the slices are thicker for speed reasons, say 10 A, that would be a very very large possible jump for atoms sitting at 5 or 15 A, etc positions.

2 - z position of atoms to a first (and second!) approximation does not matter at all really!  If you do some numerical tests using a full 3D multislice code (integrate atomic potentials in 3D) you will see that the results are very insensitive to random shifts along the beam direction.  In fact I've only really seen 1 case where these shifts matter at all - if you need quantitative 3D kikuchi bands, say in a HOLZ measurement (i.e. very high scattering angles).  So generally, you can neglect these shifts along z.

3 - 200 FP configs is a lot!  I normally recommend 16 or 32, though for some thin cases (2D materials usually) I will often use 64 or 128 FPs.  Your intuition is correct though - if you want to make the simulation more accurate, make the slices matched to your atomic cell (c/6 for example).  You can use convergence tests (i.e. compare to multislice for a few 'hard' cases like the probe directly hitting the atomic columns) to make sure your pixel size / PRISM interpolation factors are accurate enough.  And finally, run as many FPs as you can!

One final note - if you need or want enhancements like 3D potential integration and are willing to donate some development time, we'd love to implement more features!

thomasaarholt, 2018-06-09:
1. Very happy to hear that :) 
2. That's very interesting! I would not have guessed that, although perhaps this is for relatively thick samples? (in the z-direction)
3. I'm simulating on rather thin specimens (3-5nm), trying to investigate the contrast of single defects. The high number of FPs was just because I had the time for it, and wanted to remove it as a source of error for the time being. I've been spending a lot of time on putting together models without any errors in them, and I'm finally there, so I'm thinking of doing some convergence testing to see how many I need.

I'd be very keen to contribute! I'll send you an email, and see if I can find the time to help.

closed 2018-06-09 by thomasaarholt