prism-em/prismatic #33: Which side of the Potential view is the top of the sample?

thomasaarholt, 2018-06-11:
When I Calculate Potentials using the Prismatic GUI, I can use the slider to view each slice of the sample.
I have a vacancy-substitution complex on a single plane near the top of my sample (at 6Å) out of a 32Å sample, but when I view it using the slider it appears towards the higher slice numbers (slice 15-16 out of 19) (rather than at the bottom).

This seems counter-intuitive to me. Have I found a bug?

I've attached a screenshot showing the location of the substitution (heavier than the bulk atoms) in my xyz file, and the potential window showing the high value for the slides.

[image]

PS: There are two identical typos here - I'm guessing it should read "Min" and "Max".
[image]
[image]

apryor6, 2018-06-11:
Thanks for the typo notification.

As for the unexpected vacancy location, what coordinate convention are you using for the input? The coordinates are set so that the top of the sample is z=0. If you have atoms with negative coordinates, they are cyclically permuted by the cell dimension such that an atom with negative coordinates will end up "lower" than an atom at 0,0,0

thomasaarholt, 2018-06-11:
I've attached the file here. Ctrl+F "49" will get you the position of the indium atom, which is at the same depth as the vacancy. 
[ZnO_xy3_z20_d4_noDFT_Prismatic_RT.zip](https://github.com/prism-em/prismatic/files/2091967/ZnO_xy3_z20_d4_noDFT_Prismatic_RT.zip)

Are you able to spot anything odd?

cophus, 2018-07-01:
Hi Thomas,

Your cell seems to be behaving as expected.  Because all of our microscopes are "top down" i.e. gun at the top, detector at the bottom, I used the convention that the STEM probe will first go through the top of the unit cell, and propagate downwards. See this image for reference:

[image]

So if you want to recover your expected behaviour, you could flip all of the z coordinates to negative (as AJ alluded to).

cophus, 2018-07-01:
Also Thomas - I should point out that your cell is going to have some weird behaviour - you should not include large swathes of vacuum inside the unit cell (padding with vacuum on all sides for example).  The STEM probes will refract from those free surfaces, causing weird interference patterns.  

Instead, you should make the cell as close to perfectly periodic as you can - keeping the same atomic density everywhere (except around your defects) will keep the local mean inner potential roughly constant, which is closer to a real sample.

thomasaarholt, 2018-07-01:
Yikes, I'm glad you pointed that out. @apryor6 said earlier that "The coordinates are set so that the top of the sample is z=0.". That seems to contradict your figure, but your explanation solves my situation. Now I just need to reverse the axes on a bunch of figures :/

In response to your second reply: This is a "trick" that Ivan Lobato taught me at EMAT. The intention is to ensure that the sample is large enough to have sufficiently small angular resolution, without having to worry about the crystallinity of the atoms you're filling the sample with.

His opinion was that the incident probe shouldn't reach far enough out to be a problem when the ROI is such as on the image above. However, I *definitely* do not know better than you guys, so if you say it's a problem, I will correct for it. 

Just checking: If the sample was still in vacuum, but the sample was much bigger and the ROI small, would this still be a problem?

apryor6, 2018-07-01:
Slice construction involves a shift, see below. Note that the maximum z coordinate will result in `-t_z + *max_z` being zero, corresponding to the first slice.

https://github.com/prism-em/prismatic/blob/a989cd0d08bcb282a019e88eb9717b935e8a31df/src/PRISM01_calcPotential.cpp#L86-L91

thomasaarholt, 2018-07-01:
Any chance you could elaborate on your last post? It went a little over my head.

apryor6, 2018-07-01:
Sure. Colin was saying that the beam propagates down through the sample such that the top of the sample is the first slice with which there is interaction. The bit of code I linked is for assigning the atoms in the sample to an integer slice number, and that equation is setup such that larger Z coordinates will have a smaller slice number (with the limiting case that the atom with the highest z coordinate is at the shifted position 0 from the perspective of slice assignment). Thus, the atoms with high z coordinates are encountered first.

apryor6, 2018-07-01:
Apologies if my previous explanations failed to make this clear

thomasaarholt, 2018-07-20:
Thanks for the clarifying replies everyone.

I think I'll just point out that from my point of view, this is at least confusing and should be clarified in teh docs.

I think my confusion originates from assuming that the z-axis of the sample should increase alongside with the axis of the beam direction, and in the same direction of the beam.

I might have gotten further tricked by @cophus reply in #27 - it seems natural that if the beam defocus = 0 and the beam is focussed on the top of the sample, then that part of the sample would also be where z_sample = 0.

The convention is fine, but I think clear docs help the matter :) @cophus' very nice figure should make its way into the docs!

closed 2019-07-03 by lerandc

thomasaarholt, 2019-07-04:
I reckon this should remain open until this is added to the documentation. It's a pretty major aspect of the model input. I'm happy to add this when I have time.