prism-em/prismatic #34: Bugs running pyprismatic.demo

robtovey, 2018-07-19:
I installed pyprismatic via setup.py and tried to run the demo code but ran into 2 small bugs. I fixed them for myself but I don't know if these were necessary or if my hacks are wrong in some sense.
The code I was running was simply:  `python3 -c 'import pyprismatic as p; p.demo()'`

1. core.cpp did not compile into the correct place. I don't know where this ended up but adding the '--inplace' flag corrected this for me. In the end my compile line was:

        python3 setup.py build_ext --inplace install --enable-gpu

2. Running the demo also crashed when attempting to parse variables into the C++ code. I worked out that cellDimX/Y/Z were being read from the file as strings and being passed directly into core.cpp which expected floats (or type 'd'). My fix for this was to replace the line where they were read in, I don't know if this is the right place to do so. In the end params.py line 202-204 read:

        inf.readline()
        self.cellDimX, self.cellDimY, self.cellDimZ = [float(x) for x in inf.readline().split()]
        inf.close()

As I said, no longer a problem for myself but hopefully something which can help someone else.
Cheers,
Rob

apryor6, 2018-07-19:
For 1. were you trying to run the code from the source code directory? Often if adding `--inplace` was a solution this is why, so just change directories to anywhere else and try.

For 2., are the cell dimensions quoted in the input file? If you could upload the file and the command you tried to run against it that failed, I can take a look

robtovey, 2018-07-20:
For 1. Ok, yes, I'm sorry. I misread the installation guide there. It's just in the source directory it fails.

For 2. No, there are no quotes in the original file. I am running exactly the demo script which gives the printout:
[image]
This generates the parameter file (changed file extension to upload):
[temp.txt](https://github.com/prism-em/prismatic/files/2213726/temp.txt)
My python version is 3.6.4 and core.cpp was compiled with gcc/g++ 5.4.0 if that makes any difference.

closed 2018-08-14 by apryor6