prism-em/prismatic #37: pyprismatic- detectorAngleStep unit seems to be wrong

M0hsend, 2018-08-14:
I am running a CPU version of pyprismatic installed on SCARF (http://www.scarf.rl.ac.uk/home). When I set the pr.Metadata.save3DOutput to True, with pr.Metadata.probeSemiangle set to 30 and pr.Metadata.detectorAngleStep to 1, I expect to get a 3D object with the detector plane divided into 30 bins 1 mrad each. In practice, I get only 1 bin. In order to get 30 bins on detector plane, pr.Metadata.detectorAngleStep needs to be set to 0.003. 

Is the unit for this parameter not, as stated in docs, in mrads?

apryor6, 2018-08-14:
That's a bug. This is similar to [this issue](https://github.com/prism-em/prismatic/issues/30)

M0hsend, 2018-08-15:
Many thanks @apryor6

lerandc, 2019-07-05:
Bug is on its way to correction-- see issue #51 for details. Update has not made its way to master branch as of yet.

closed 2021-06-16 by lerandc