prism-em/prismatic #39: Difference between GUI and Pyprismatic Outputs

M0hsend, 2018-08-22:
I am running similar sim tests (close to what is described in the tutorial here: http://prism-em.com/tutorialSuperSTEM/), comparing the 3D outputs from GUI versus pyprismatic. These are the two parameter files:
[py_test_param.txt](https://github.com/prism-em/prismatic/files/2310896/py_test_param.txt)
[GUI_test_param.txt](https://github.com/prism-em/prismatic/files/2310901/GUI_test_param.txt)
I am using the same crystal model as the [tutorial](http://prism-em.com/data/SuperSTEM/barium_neodymium_titanate_0_1_0.xyz). 

The 3D outputs (sum-ed over detector bins) are as follows:
[image]

The order of the bins seem to be reversed, but also the intensity values are different.

Here are BF and HAADF images from the two sims (summing over similar bins in the two cases):
[image]
[image]

What is causing the difference between the two simulations?

apryor6, 2018-08-22:
This is most likely related to [this issue](https://github.com/prism-em/prismatic/issues/30). There are a few parameters that are currently improperly converted between rad/mrad between pyprismatic and prismatic.

M0hsend, 2018-08-22:
Many thanks for your prompt reply @apryor6 
I think I have that covered for the detector angles.
This is what I have for GUI:
`--detector-angle-step:1`
and this for the pyprismatic sim:
`meta.detectorAngleStep=1e-3 #angular step size for detector integration bins (in mrad)`
And this is to get same number of bins in the 3D output.

I am just trying to match the 3D outputs of the two routes. I would much appreciate it if you could kindly suggest any changes in metadata to achieve that.

lerandc, 2019-07-05:
It has been a good while @M0hsend , but this does seem to be related to a conversion issue--specifically, based on the two parameter files, it was likely the probe semiangle that caused the issue. Hopefully this is fixed now- see issue #51 for details.

closed 2021-06-16 by lerandc