prism-em/prismatic #41: 4D STEM Diffraction image calibration

thomasaarholt, 2018-11-01:
I am using Pyprismatic.

What is the pixel size of the 4D STEM output? And what factors does it depend on? The pixel size appears to vary if I change interpolation factor, but not if I change `meta.detectorAngleStep`.

HamishGBrown, 2019-04-17:
Hi Thomas, see section 4.1 Sampling of Earl Kirkland's "Advanced Computing in Electron Microscopy" for a detailed explanation. In multislice if your cell dimensions are X \times Y then your pixel size in reciprocal (diffraction) space will be 1/X \times 1/Y.

Using a PRISM interpolation factor _f_ complicates this somewhat, here the exit surface wave is only calculated for every _f_ diffraction space pixel. So your pixel size in diffraction space is actually 1/X \times 1/Y.

lerandc, 2019-07-03:
Dimension data is now written to the output automatically in the new HDF5 output, so now there should be directly accessible data regarding the pixel size.

closed 2019-07-03 by lerandc

thomasaarholt, 2019-07-04:
Super stuff! Excited to check it out!