prism-em/prismatic #43: Ram use on supercomputer

thomasaarholt, 2018-12-19:
Do any of you have experience running on a supercomputer? Ours users the SLURM system.

I am very unsure of how to calculate/choose the memory, CPU and GPU settings for running a job. 

I'm trying to run a relatively large sample (dimensions 72x78x32 Å3) with a defect in the centre. With interpolation factors of 2 I hit a memory error as the job uses more than 32 GB of memory. Interpolation of 4 seems to run successfully, but I'd like to check how different the results are.

On our slurm system we can choose number of tasks, tasks per node, cpus per task and memory per cpu. It's already confusing, but additionally I don't know how prismatics code fits into their system.

For testing purposes I've been using:

```
#SBATCH --ntasks=5
#SBATCH --cpus-per-task=16
#SBATCH --mem-per-cpu=3000
```
Any advice at all is appreciated.

thomasaarholt, 2018-12-20:
Thanks for an excellent answer! I'll give this a shot and see how it goes!

thomasaarholt, 2019-01-02:
Well, it looks like running on the supercomputer, either with 16 CPUs or with the two K20 GPUs, is significantly slower than the single GTX1080TI card we have. Unfortunately that computer "only" has 32 GB ram, and it looks like my simulation requires more than that. Are there any smart "memory management" solutions that I should be aware of?

thomasaarholt, 2019-01-03:
Trying it now, but it seems to still eat up all my ram.
[image]
According to GPU and CPU monitoring tools, CPU and GPU is barely being used.

This is all via pyprismatic, and I do get confirmation from the terminal that it is running it streamed.  While I was using "auto" previously, it was definitely using the single transfer method before.

apryor6, 2019-01-03:
You're correct that your limitation is memory. The CPU and GPU utilization is low because you're machine is too busy struggling to handle the memory bottleneck. In this situation you either need to adjust the simulation settings to require more RAM or get a larger machine. 32GB is not a lot of RAM by today's standards.. you can spin up a virtual machine on a cloud platform like AWS that has 128GB or more for a few dollars per hour or build a similar spec workstation in-house.