prism-em/prismatic #51: Crashes and wrong output with numThreads=0 and alsoDoCPUWork=False

thomasaarholt, 2019-02-27:
I wanted to run simulations only on the GPU, so I set numThreads=0 and alsoDoCPUWork=False. This resulted in the following image:
[image]

With those arguments as default values, the image looks normal:
[image]

Additionally, if one just sets numThreads=0, the simulation dies with no error very early:
```
Entering PRISM01_calcPotential
Waiting for threads...
Launching GPU worker on stream #0 on GPU #0
Launching GPU worker on stream #1Computing Probe Position #0 on GPU #0/
5250Launching GPU worker on stream #2
 on GPU #0
Computing Probe Position #100/5250
# software dies
C:\Users\Me\Documents\temp2\cube_Bulk_RT
```

thomasaarholt, 2019-02-27:
I'm guessing that the numThreads are needed to compute the projected potential slices, but never gets them.

HamishGBrown, 2019-04-17:
Hey Thomas, the potentials are _only_ calculated on the CPU so if you are able to set the variable pars.meta.numThreads = 0 then no potential will be calculated. This is why the GUI won't let you do this:

`[void PRISMMainWindow::setNumThreads(const int& num){
    if (num > 0){
        this->meta->numThreads = num;
        std::cout << "Setting number of CPU Threads to " << num << std::endl;
        QMutexLocker gatekeeper(&dataLock);
        this->pars.meta.numThreads = num;
    }
}](https://github.com/prism-em/prismatic/blob/0a29fa5571d49bb65bc650bd2f5256588fac9abc/Qt/prismmainwindow.cpp#L697)`

Only if the number of threads > 0 will the variable be updated.

I'm working on a fix where if you set number of threads to 0 then the flag alsoDoCPUWork will be set to false but numThreads will be set to 1.

lerandc, 2019-07-03:
To clarify, Thomas, is this through PyPrismatic?

thomasaarholt, 2019-07-04:
Yes, definitely through pyprismatic, but I think I tested it with the CLI version as well. Unsure.

lerandc, 2019-07-05:
Recently tested this in the CLI and in a build of pyprismatic. The issue seems to have been related due to the fact that the core module for pyprismatic did not actually parse the inputs for validity (nor performed any unit conversions, ie, potential sources for issues #50, #39, #37). I just recently added a check into the core module which runs the calculation, as well as the unit fixes (see [core.cpp](https://github.com/prism-em/prismatic/commit/ca0b56a3cf4dabb74a80f3d1492412d86a8aa173) for details).

I did also get a version of the setup script to link the HDF5 libraries correctly, so @thomasaarholt if you want to try the new version of prismatic out and see if you find the same issues still, hopefully it should all work. Pip install still refers to an old archival version, but, if you build prismatic regularly with GPU enabled and go to the main source directory and run the following, it seems to build stably.  Of course, replace the library and include directories according to local installation paths. If you set environment variables to install prismatic and do this right after, that might work as well.
```
python3 setup.py build_ext --include-dirs=/usr/local/boost_1_70_0/:/usr/local/cuda/include/:/usr/local/hdf/include/ --library-dirs=/usr/local/cuda/lib64/:/usr/local/hdf/lib/ install --enable-gpu
```

closed 2021-06-16 by lerandc