prism-em/prismatic #53: GUI: Save projected potentials has no effect

thomasaarholt, 2019-03-06:
As far as I can tell, the "Save projected potentials" option in the GUI version of prismatic does nothing at all.

HamishGBrown, 2019-04-17:
Hey Thomas, it will save a file called "potential.mrc" in the run directory for the linux build. It might not work on the Windows version, I'll test this once I compile on my laptop.

In the fix I'll shortly be uploading the program takes the output filename and replaces the ending with "_potential.mrc".

lerandc, 2019-07-03:
This should be fixed in the recent release- Prismatic will now save the projected potentials into an HDF5 along with all of the other requested output.

yobc401, 2019-08-12:
@lerandc  I installed the latest version of Prismatic codes in my linux system, but I'm not able to find any file regarding "Save projected potentials" options although I turn on the option like "meta.savePotentialSlices = True". ("potential.mrc" file is never founded after running pyprismatic code) Could you let me know why the file isn't made?

lerandc, 2019-08-16:
In the new version of prismatic, all outputs are now contained within a single output file in the HDF5 format. Can you confirm that a file "fileNameHere.h5" exists in your output directory? The potential slice data should be contained in the path "/4DSTEM_simulation/data/realslices/ppotential/realslice/" as a 3D array with dimensions "rx,ry,slice".

closed 2021-06-16 by lerandc