prism-em/prismatic #55: ASE integration for xyz files

SBW90, 2019-04-26:
I noticed that xyz files produced in the python library atomic simulation environment (ASE) are not compatible with prismatic. I've written a script that converts "ASE_xyz" to "prismatic_xyz" compatible files, its a bit ugly but might be useful to others. I'm new to git so not sure how to work propose working this in.

thomasaarholt, 2019-04-26:
That's nice! I've written a similar script for home-use. How do you
incorporate the Debye waller (rms) factor?

thomasaarholt, 2019-06-08:
@SBW90 did you publish your code somwhere?

Eoghan42, 2019-07-08:
That'd be great to have alright! I've written a similar set of functions. For example, creating a prismatic .xyz and .cif from Atomap sublattices, as well as converting vesta .xyz into prismatic .xyz. 

Where, do you think, would be the best place for these (if they don't fit into Atomap)? Is there a "babelfish" type program for converting between formats in python?

About the rms values, is importing from a library of calculated values an option?

thomasaarholt, 2019-07-08:
Since we are accumulating a number of scripts that solve slightly different problems, I suggest that we stick them in a github repo and label them clearly in a readme.

I've written a script to write ASE files to [prismatic xyz and multem matlab files](def save_cell(cell, name, path="", only_save=False, temperature='LN2'):).

RMS values can come from a high number of sources. DFT often supplies them, but that depends on the DFT framework one is using. What did you have in mind?

ericpre, 2019-07-08:
ASE is a kind of babelfish, since it can read and write [many supercell file format](https://wiki.fysik.dtu.dk/ase/ase/io/io.html#ase.io.write) and to my opinion, the best place to write the IO plugin would be in ASE itself. I did it for muSTEM a few years ago, see https://gitlab.com/ase/ase/blob/master/ase/io/mustem.py#L63.

To add a new IO format to ASE, read https://wiki.fysik.dtu.dk/ase/ase/io/io.html#adding-a-new-file-format-to-ase

thomasaarholt, 2019-07-08:
@ericpre That's nice! I thought it would be a problem that one needed DW values, but that is good to see isn't an issue!

alex-rakowski, 2021-06-14:
Closing as [ASE](https://wiki.fysik.dtu.dk/ase/ase/io/formatoptions.html?highlight=ase%20io%20prismatic#prismatic) can read and write to prismatic style xyz.

closed 2021-06-14 by alex-rakowski