prism-em/prismatic #60: Using PyPrismatic with mybinder.org

Eoghan42, 2019-07-08:
Hey everyone! 
Is it possible to use PyPrismatic with [binder ](https://gke.mybinder.org/)? Has anyone tried? I assumed there would be issues with the dependencies FFTW and Boost. This is clear below. However, I know little about how these dependencies are compiled. Let me know if this is a question that would be better suited for the binder people, or if it's a bit daft.

I recently tried to create a binder of a Jupyter Notebook with PyPrismatic one of the imported packages in the requirements.txt. It failed with this error:

```
gcc -pthread -B /srv/conda/envs/notebook/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -f
wrapv -O3 -Wall -Wstrict-prototypes -fPIC -I./include -I/srv/conda/envs/notebook/include/python3.7m -c src
/go.cpp -o build/temp.linux-x86_64-3.7/src/go.o -std=c++11
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    In file included from ./include/Multislice_calcOutput.h:26:0,
                     from ./include/Multislice_entry.h:20,
                     from src/go.cpp:19:
    ./include/utility.h:24:10: fatal error: fftw3.h: No such file or directory
     #include "fftw3.h"
              ^~~~~~~~~
    compilation terminated.
    error: command 'gcc' failed with exit status 1
```

Anyone know where this `#include "fftw3.h"` should be placed in the binder, or if it's possible at all for binder to deal with it?

thomasaarholt, 2019-07-08:
I'm pretty sure noone has tried, but this is a fun thing to try.

I haven't tried using Binder in general myself, but could try adding a binder dependency for Fftw, although I'm not certain if that will be properly registered by gcc.

Eoghan42, 2019-07-09:
Thanks for the quick reply! pyFFTW isn't an option is it? As far as I understand, prism uses fftw in the background when doing calculations in C++? 

Perhaps I should try to use c++ in Jupyter and Binder and see how far I can get with it.

thomasaarholt, 2019-07-09:
It definitely uses the c++ implementation, I was just thinking that maybe you'd be able to find the same file via a conda/pip install. As said, I'm not sure how Binder works, but if you're able to just add in the files and find a way for gcc to be aware of them, it might work.

Does binder "exist" on Windows as well as linux? I've got [compiled files here if you fancy them.](https://github.com/prism-em/prismatic/issues/7#issuecomment-344869157)

fomightez, 2019-08-09:
@PinkShnack According to [here](https://anaconda.org/conda-forge/fftw), conda can install `fftw` and so for using it with MyBinder, you should switch from using a `requirements.txt` file to using an `environment.yml` file. `environment.yml` gets used by conda. Conda has most of the pip installable items too; however, in cases where it doesn't overlap you can move your `pip` requirements there too, like I did [here](https://github.com/fomightez/circos-binder/blob/master/environment.yml).

ericpre, 2020-03-29:
pyprismatic can be installed from conda-forge:

```
conda install pyprismatic -c conda-forge
```

fomightez, 2020-03-29:
@ericpre Sorry, I should have pointed out, not everything has to be converted to using conda. `environment.yml` files are also easily able to specify pip installs along with conda.  If the preference is to have pip install pyprismatic as the documentation suggests, you can stick with that using `environment.yml`.  (One reason for favoring one over is that maybe one tends to get updated faster.)  [Here](https://github.com/fomightez/jupyter-desktop-server/blob/master/environment.yml) is an example of using that. And if you did still want to use `conda-forge` as a channel, you can; [here](https://github.com/binder-examples/conda/blob/master/environment.yml) is an example of that.

PinkShnack, 2020-07-09:
> pyprismatic can be installed from conda-forge:
> 
> ```
> conda install pyprismatic -c conda-forge
> ```

@ericpre This is fantastic for Prismatic, thanks a lot for doing all that work.
Shouldn't this link be on the Prismatic website? https://github.com/conda-forge/prismatic_split-feedstock

And to confirm, including prismatic in the `environment.yml` now successfully works with Binder. Very cool!
Example:

```
channels:
  - defaults
  - conda-forge
dependencies:
  - python=3.7
  - prismatic
```

Will close this issue in a couple of days.

lerandc, 2020-07-09:
@PinkShnack yes, noted! added to issues for the website repo prism-em/prism-em.github.io#7. we plan to update the website on the (incoming) next release of prismatic

closed 2020-07-10 by PinkShnack