prism-em/prismatic #67: Issue in compiling prismatic using GCC version 4.8

Quantumstud, 2019-08-15:
**I am trying to compile to the Prismatic source code. I am having issues compiling using the gcc-4.8 ** 
--------------------------------------------------------------------------------------
Scanning dependencies of target prismatic
[ 13%] Building CXX object CMakeFiles/prismatic.dir/src/configure.cpp.o
[ 13%] Building CXX object CMakeFiles/prismatic.dir/src/driver.cpp.o
[ 20%] Building CXX object CMakeFiles/prismatic.dir/src/PRISM02_calcSMatrix.cpp.o
[ 26%] Building CXX object CMakeFiles/prismatic.dir/src/PRISM03_calcOutput.cpp.o
[ 33%] Building CXX object CMakeFiles/prismatic.dir/src/WorkDispatcher.cpp.o
[ 40%] Building CXX object CMakeFiles/prismatic.dir/src/PRISM01_calcPotential.cpp.o
[ 46%] Building CXX object CMakeFiles/prismatic.dir/src/Multislice_calcOutput.cpp.o
[ 53%] Building CXX object CMakeFiles/prismatic.dir/src/PRISM_entry.cpp.o
[ 60%] Building CXX object CMakeFiles/prismatic.dir/src/Multislice_entry.cpp.o
[ 66%] Building CXX object CMakeFiles/prismatic.dir/src/projectedPotential.cpp.o
[ 73%] Building CXX object CMakeFiles/prismatic.dir/src/parseInput.cpp.o
[ 80%] Building CXX object CMakeFiles/prismatic.dir/src/utility.cpp.o
[ 86%] Building CXX object CMakeFiles/prismatic.dir/src/atom.cpp.o
[ 93%] Building CXX object CMakeFiles/prismatic.dir/src/go.cpp.o
[100%] Linking CXX executable prismatic
CMakeFiles/prismatic.dir/src/PRISM01_calcPotential.cpp.o: In function `Prismatic::PRISM01_calcPotential(Prismatic::Parameters<float>&)':
PRISM01_calcPotential.cpp:(.text+0x2911): undefined reference to `H5::H5Location::createGroup(std::string const&, unsigned long) const'
PRISM01_calcPotential.cpp:(.text+0x300b): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
PRISM01_calcPotential.cpp:(.text+0x301d): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
PRISM01_calcPotential.cpp:(.text+0x3032): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
PRISM01_calcPotential.cpp:(.text+0x311f): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
PRISM01_calcPotential.cpp:(.text+0x3134): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
CMakeFiles/prismatic.dir/src/PRISM01_calcPotential.cpp.o:PRISM01_calcPotential.cpp:(.text+0x3149): more undefined references to `H5::Attribute::write(H5::DataType const&, std::string const&) const' follow
CMakeFiles/prismatic.dir/src/PRISM01_calcPotential.cpp.o: In function `Prismatic::PRISM01_calcPotential(Prismatic::Parameters<float>&)':
PRISM01_calcPotential.cpp:(.text+0x32d7): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
PRISM01_calcPotential.cpp:(.text+0x3542): undefined reference to `H5::H5Location::createDataSet(std::string const&, H5::DataType const&, H5::DataSpace const&, H5::DSetCreatPropList const&) const'
PRISM01_calcPotential.cpp:(.text+0x36af): undefined reference to `H5::H5Location::openDataSet(std::string const&) const'
CMakeFiles/prismatic.dir/src/PRISM_entry.cpp.o: In function `Prismatic::PRISM_entry(Prismatic::Metadata<float>&)':
PRISM_entry.cpp:(.text+0x330e): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
PRISM_entry.cpp:(.text+0x3965): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
PRISM_entry.cpp:(.text+0x3a3d): undefined reference to `H5::H5Location::openDataSet(std::string const&) const'
PRISM_entry.cpp:(.text+0x43cb): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
PRISM_entry.cpp:(.text+0x4406): undefined reference to `H5::H5Location::openDataSet(std::string const&) const'
CMakeFiles/prismatic.dir/src/Multislice_entry.cpp.o: In function `Prismatic::Multislice_entry(Prismatic::Metadata<float>&)':
Multislice_entry.cpp:(.text+0x2fa4): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
Multislice_entry.cpp:(.text+0x3dea): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
Multislice_entry.cpp:(.text+0x3e5a): undefined reference to `H5::H5Location::openDataSet(std::string const&) const'
Multislice_entry.cpp:(.text+0x47d9): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
Multislice_entry.cpp:(.text+0x481f): undefined reference to `H5::H5Location::openDataSet(std::string const&) const'
CMakeFiles/prismatic.dir/src/utility.cpp.o: In function `Prismatic::setup2DOutput(Prismatic::Parameters<float>, unsigned long, float)':
utility.cpp:(.text+0x1f8a): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
utility.cpp:(.text+0x1f9d): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
utility.cpp:(.text+0x2046): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
utility.cpp:(.text+0x2059): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
CMakeFiles/prismatic.dir/src/utility.cpp.o: In function `Prismatic::setup2DOutput(Prismatic::Parameters<float>, unsigned long, double)':
utility.cpp:(.text+0x2c2a): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
CMakeFiles/prismatic.dir/src/utility.cpp.o:utility.cpp:(.text+0x2c3d): more undefined references to `H5::Attribute::write(H5::DataType const&, std::string const&) const' follow
CMakeFiles/prismatic.dir/src/utility.cpp.o: In function `Prismatic::writeStringArray(H5::DataSet, std::string*, unsigned long long)':
utility.cpp:(.text+0x34ab): undefined reference to `H5::DataSet::write(std::string const&, H5::DataType const&, H5::DataSpace const&, H5::DataSpace const&, H5::DSetMemXferPropList const&) const'
CMakeFiles/prismatic.dir/src/utility.cpp.o: In function `Prismatic::setupDPCOutput(Prismatic::Parameters<float>, unsigned long, float)':
utility.cpp:(.text+0x3afd): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
utility.cpp:(.text+0x3b0d): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
utility.cpp:(.text+0x3bb1): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
utility.cpp:(.text+0x3bc4): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
CMakeFiles/prismatic.dir/src/utility.cpp.o: In function `Prismatic::setupDPCOutput(Prismatic::Parameters<float>, unsigned long, double)':
utility.cpp:(.text+0x49f0): undefined reference to `H5::Attribute::write(H5::DataType const&, std::string const&) const'
CMakeFiles/prismatic.dir/src/utility.cpp.o:utility.cpp:(.text+0x4a00): more undefined references to `H5::Attribute::write(H5::DataType const&, std::string const&) const' follow
CMakeFiles/prismatic.dir/src/utility.cpp.o: In function `Prismatic::writeDatacube4D(Prismatic::Parameters<float>, float*, unsigned long long const*, unsigned long long const*, float, std::string)':
utility.cpp:(.text+0xcd56): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
CMakeFiles/prismatic.dir/src/utility.cpp.o: In function `Prismatic::writeDatacube4D(Prismatic::Parameters<float>, double*, unsigned long long const*, unsigned long long const*, double, std::string)':
utility.cpp:(.text+0xd0e6): undefined reference to `H5::H5Location::openGroup(std::string const&) const'
collect2: error: ld returned 1 exit status
CMakeFiles/prismatic.dir/build.make:451: recipe for target 'prismatic' failed
make[2]: *** [prismatic] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/prismatic.dir/all' failed
make[1]: *** [CMakeFiles/prismatic.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
-----------------------------------------------------------------------------------------------------------------------
**Ironically when I change the gcc version to 7.4.0. It is built. However, this gcc cannot compile the GPU version.**
---------------------------------------------------------
(base) tara@tara-XPS-8930:~/Downloads/STEMSoftwares/Simulation/PRISM/prismatic-master/build$ make -j 8
Scanning dependencies of target prismatic
[ 13%] Building CXX object CMakeFiles/prismatic.dir/src/PRISM01_calcPotential.cpp.o
[ 13%] Building CXX object CMakeFiles/prismatic.dir/src/WorkDispatcher.cpp.o
[... cut ...]
[ 66%] Building CXX object CMakeFiles/prismatic.dir/src/parseInput.cpp.o
[ 73%] Building CXX object CMakeFiles/prismatic.dir/src/projectedPotential.cpp.o
[ 80%] Building CXX object CMakeFiles/prismatic.dir/src/utility.cpp.o
[ 86%] Building CXX object CMakeFiles/prismatic.dir/src/atom.cpp.o
[ 93%] Building CXX object CMakeFiles/prismatic.dir/src/go.cpp.o
[100%] Linking CXX executable prismatic
[100%] Built target prismatic
-------------------------------------------------------------------------------------------------------------------

Quantumstud, 2019-08-16:
Hi! 
I found the problem was with the HDF5 installation. CMake was not able to find the libraries installed. The HDF5 libraries have to be re-compiled using the gcc-4.8. The libraries compiled with gcc-7 won't work with gcc-4.8 which is why gcc-7 worked, and gcc-4.8 didn't. So I recompiled HDF5 using the gcc-4.8.

### Just an additional suggestion CMake might often fail to find the HDF5 libraries One way would be setting the environment variable to set the library location of the HDF5 installation. However, the more natural way is to use the following command during the configuration of HDF5 package. 

Go to the top directory of the downloaded HDF5 folder.
`./configure --prefix=/usr/local --enable-shared --enable-hl --enable-threadsafe--enable-cxx --enable-unsupported
make -j 2
sudo su
make install`

This will install both the static and shared libraries in the folders which can be accessed by CMake easily and helps us to save the tiring process of setting the environment variable.

I hope this helps anyone who faces the similar compilation issue.

lerandc, 2019-08-16:
Thanks for passing along your solution to the issue! I will add a note to the main website that clarifies this compilation issue.

apryor6, 2019-08-16:
Thanks for sharing resolution.

closed 2019-08-16 by apryor6