prism-em/prismatic #70: macOS GUI Crashes

sezelt, 2019-12-12:
The GUI loads and will generate the potentials, but when I try to generate a probe it crashes (all settings default):

```
Process:               prismatic-gui [16776]
Path:                  /Users/USER/*/prismatic-gui.app/Contents/MacOS/prismatic-gui
Identifier:            ???
Version:               ??? (0.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           prismatic-gui [16776]
User ID:               501

Date/Time:             2019-12-11 20:10:15.006 -0800
OS Version:            Mac OS X 10.15.1 (19B88)
Report Version:        12
Anonymous UUID:        513A3165-6E70-019E-0244-FF7A7946F176

Time Awake Since Boot: 470000 seconds

System Integrity Protection: enabled

Crashed Thread:        5  ProbeThread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [16776]

VM Regions Near 0:
--> 
    __TEXT                 000000010cead000-000000010d3b0000 [ 5132K] r-x/rwx SM=COW  /Users/USER/*/prismatic-gui.app/Contents/MacOS/prismatic-gui

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	0x00007fff642be989 _platform_bzero$VARIANT$Haswell + 41
1   libsystem_malloc.dylib        	0x00007fff64296026 medium_malloc_should_clear + 244
2   libsystem_malloc.dylib        	0x00007fff6427fb22 szone_malloc_should_clear + 182
3   libsystem_malloc.dylib        	0x00007fff64280e7b malloc_zone_calloc + 99
4   libsystem_malloc.dylib        	0x00007fff64280dfb calloc + 24
5   com.apple.CoreGraphics        	0x00007fff2d09a988 RIPLayerCreate + 362
6   com.apple.CoreGraphics        	0x00007fff2d0a6a48 ripc_BeginLayer + 1255
7   com.apple.CoreGraphics        	0x00007fff2d0a623a CGContextBeginTransparencyLayerWithRect + 85
8   com.apple.CoreGraphics        	0x00007fff2d0c438a CGContextBeginTransparencyLayer + 172
9   com.apple.QuartzCore          	0x00007fff384911c0 -[CALayer _renderForegroundInContext:] + 2972
10  com.apple.AppKit              	0x00007fff2a0b14dd -[_NSViewBackingLayer _renderForegroundInContext:] + 155
11  com.apple.QuartzCore          	0x00007fff3848fbfe -[CALayer renderInContext:] + 1649
12  com.apple.QuartzCore          	0x00007fff38491e29 -[CALayer _renderSublayersInContext:] + 528
13  com.apple.QuartzCore          	0x00007fff3848fc0d -[CALayer renderInContext:] + 1664
14  com.apple.AppKit              	0x00007fff2a0b1284 -[NSView(NSInternal) _renderLayerInContext:] + 292
15  com.apple.AppKit              	0x00007fff2a67bec6 -[NSView(NSInternal) _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 851
16  com.apple.AppKit              	0x00007fff2a0b0cca -[NSView displayRectIgnoringOpacity:inContext:] + 845
17  libqmacstyle.dylib            	0x0000000114a00be9 0x1149f4000 + 52201
18  ???                           	0x0000600000cb8210 0 + 105553129603600

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff642c956c start_wqthread + 0

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff64209166 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff642096cc mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff2cbe736b __CFRunLoopServiceMachPort + 322
3   com.apple.CoreFoundation      	0x00007fff2cbe6907 __CFRunLoopRun + 1695
4   com.apple.CoreFoundation      	0x00007fff2cbe5fe3 CFRunLoopRunSpecific + 499
5   com.apple.AppKit              	0x00007fff29df9dba _NSEventThread + 132
6   libsystem_pthread.dylib       	0x00007fff642ccd36 _pthread_start + 125
7   libsystem_pthread.dylib       	0x00007fff642c958f thread_start + 15

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff642c956c start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff642c956c start_wqthread + 0

Thread 5 Crashed:: ProbeThread
0   ???                           	000000000000000000 0 + 0
1                                 	0x000000010ced43e7 Prismatic::PRISM02_calcSMatrix(Prismatic::Parameters<float>&) + 391
2                                 	0x000000010cf50f4e ProbeThread::run() + 1646
3   libQt5Core.5.12.5.dylib       	0x000000010e2b83d0 0x10e2a0000 + 99280
4   libsystem_pthread.dylib       	0x00007fff642c958f thread_start + 15

Thread 5 crashed with X86 Thread State (64-bit):
  rax: 0x000000010d3b7470  rbx: 0x00007fda61fb0290  rcx: 0x0000000000000001  rdx: 0x0000000000000001
  rdi: 0x0000700007b1f580  rsi: 0x00000000000120a8  rbp: 0x0000700007b1e060  rsp: 0x0000700007b1e028
   r8: 0x00000000000130a8   r9: 0x0000000000000000  r10: 0x00006000032ad330  r11: 0x00006000032ad328
  r12: 0x0000000000000000  r13: 0x00007fda61e6db50  r14: 0x0000700007b1f580  r15: 0x000000000000000a
  rip: 0x0000000000000000  rfl: 0x0000000000010202  cr2: 0x0000000000000000
  
Logical CPU:     2
Error Code:      0x00000014 (no mapping for user instruction write)
Trap Number:     14

```

sezelt, 2019-12-12:
This happens with the conda-forge build btw

ericpre, 2019-12-12:
@sezelt if you think this is related to the packaging of prismatic on conda-forge, it may be appropriate to discuss this in the corresponding feedstock: https://github.com/conda-forge/prismatic_split-feedstock.

sezelt, 2019-12-13:
I've now been able to duplicate in on a Mac build from source and the Windows MSI installer. It occurs when you hit the "Calculate" button in the probe analyzer while the Algorithm selector is set to Multislice.