prism-em/prismatic #75: The simulation metadata are empty

ericpre, 2020-03-13:
The simulation metadata are supposed to be saved in the group `/4DSTEM_experiment/metadata/metadata_0/original/simulation_parameters` but all the key-value pairs are empty!

To repoduce:
```
prismatic -i SI100.XYZ
```

```python
import h5py

f = h5py.File('output.h5', 'r')
md = f.get('4DSTEM_simulation/metadata/metadata_0/original/simulation_parameters')

print(md)
<HDF5 group "/4DSTEM_simulation/metadata/metadata_0/original/simulation_parameters" (0 members)>
```

lerandc, 2020-05-20:
Hi @ericpre, it looks like a bit of semantics confusion with h5py. The metadata are stored in the group attributes, so to access them `md.attrs`, which can be listed out by 'md.attrs.keys()'

for self-reference/documentation updating:
List all simulation parameters and values (h5py method):
`import h5py
fp = "output.h5"
f = h5py.File(fp, 'r')
md = f.get('4DSTEM_simulation/metadata/metadata_0/original/simulation_parameters')
for key in md.attrs.keys():
    print(key, md.attrs[key]) #prints string values as byte strings`

ericpre, 2020-05-20:
Yes, I think that I misread the documentation that day and miss the attributes bit, this is indeed working fine (https://github.com/hyperspy/hyperspy/pull/2396/files#diff-9a9e452522a6c34ae2dd3782a26581d2R719-R735), forgot to close this issue - sorry for the noise!

ericpre, 2020-05-20:
While you are here, could you please review some of the open PRs (when you have time)? :)

closed 2020-05-20 by ericpre

lerandc, 2020-05-20:
Yes, certainly!