prism-em/prismatic #79: Feature request: Frozen phonon pass # during multislice?

stevenspurgeon, 2020-05-20:
I'd like to see if it's possible to include the frozen phon pass # during full multislice calculations, either in the progress bar or the command line. It would be useful for larger, more time-intensive simulations to see how far along you are. Thank you.

lerandc, 2020-05-20:
Hi @stevenspurgeon, this should there on the command line version at least. When it prints out the simulation parameters, it should print the frozen phonon pass (indexed from 0) first. For example, on the 4th Frozen Phonon, it will print something similar to:

"extracted 8 atoms from 10 lines in SI100.XYZ
Warning: User requested 4 GPUs but only 1 were found. Proceeding with 1.
deviceProperties.major = 6
deviceProperties.maxThreadsPerBlock = 1024
targetNumBlocks = 160
Frozen Phonon #3

Simulation parameters:
...."

stevenspurgeon, 2020-05-20:
Hi @lerandc, yeah it seems right now that it's not showing up in the command line. Another great option would be if we could see it in the progress bar, say something like "Computing final output (Multislice / FP = 1)." Thank you!

lerandc, 2020-05-20:
Hmm, interesting, can you describe a bit more what you mean? Are you running the GUI from the command line, or the pure prismatic executable itself?