prism-em/prismatic #83: HDF5 3D Output has mrad scale off by a factor of 1000

thomasaarholt, 2020-08-21:
As seen from the x-axis in the following spectrum (after loading the prismatic output with hyperspy (using the `.emd` suffix), the scale appears to be off by a factor of 1000. 0.02 mrad should be 20 mrad here.

I found [references to "mrad" in the code](https://github.com/prism-em/prismatic/blob/60b754e75949e7d4da79b912d4dfdec880880deb/src/utility.cpp#L627), but couldn't work out where the scale is saved.

[image]

thomasaarholt, 2020-08-27:
@ericpre I notice that you are currently updating the conda package. I just tried with the pyprismatic package that was uploaded to conda-forge 3 days ago (conda reads it as v. 1.2.1), and get the following very small mrad scale when I've given it 1 mrad. I don't quite know if what's in the `dev` branch is the same as on conda, but I thought I'd mention it.

[image]

ericpre, 2020-08-28:
The packages on conda-forge are 1.2.1 as it clearly says and it has nothing to do with the development branch. This bug is present in the 1.2.1 release, so it is also present in the corresponding packages. Since it has been fixed in #80, this issue should be closed and the solution at the moment is to wait for the next release or install the development branch. There is now a conda recipe in the `dev` branch which can help building from source.

thomasaarholt, 2020-08-28:
Thanks. The reason I was (incorrectly) wondering if they were the same, is that the master branch setup.py refers to version 1.2.0. What branch is the code for 1.2.1 in? 

Thanks for the work you've been doing with getting the Linux gpu version on conda. I'll take a look at the conda recipe in the dev branch and see if I can manage to compile it myself.

closed 2020-08-28 by thomasaarholt

ericpre, 2020-08-28:
> Thanks. The reason I was (incorrectly) wondering if they were the same, is that the master branch setup.py refers to version 1.2.0. What branch is the code for 1.2.1 in?

The package pull the github release (and therefore a specific git tag), it doesn't use `setup.py`:

https://github.com/conda-forge/prismatic_split-feedstock/blob/master/recipe/meta.yaml#L15

thomasaarholt, 2020-08-28:
> There is now a conda recipe in the `dev` branch which can help building from source.

Thanks! This worked great!
In case anyone else tries, to build and install the gpu version, there were a couple of environment variables that weren't set on our system:
`export CUDA_HOME=/usr/local/cuda-10.1` # your equivalent of
`export PYTHON=/home/thomasaar/.conda/envs/py3/bin/python` # your equivalent of
 from `prismatic/`call `recipe/build_pyprismatic.sh gpu`.

ericpre, 2020-08-28:
The conda recipe is supposed to be used with conda build (https://docs.conda.io/projects/conda-build).

thomasaarholt, 2020-08-28:
I started with that, and got a few error messages that I wasn't entirely sure how to resolve:
```
# in prismatic/
conda build . # or conda build recipe or conda build recipe/meta.yaml
...
Attempting to finalize metadata for prismatic_cli
INFO:conda_build.metadata:Attempting to finalize metadata for prismatic_cli
Error: Failed to render jinja template in /home/thomasaar/prismatic/recipe/meta.yaml:
'cuda_compiler_version' is undefined
```
Then I tried the `./build_pyprismatic.sh` without `gpu`, got that working, and then fixed the errors for `./build_pyprismatic.sh gpu` and was happy enough :)