prism-em/prismatic #110: Error installing GPU version of pyprismatic

drpaulsharp, 2021-07-20:
I want to install the GPU version of pyprismatic on a linux machine. I am able to install the standard CPU version using pip, but when I try to `pip install pyprismatic --install-option="--enable-gpu"`, I get the following error:

```
    ERROR: Command errored out with exit status 1:
     command: /home/paul/env/images/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-takbc5vy/pyprismatic/setup.py'"'"'; __file__='"'"'/tmp/pip-install-takbc5vy/pyprismatic/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-sugpug87/install-record.txt --single-version-externally-managed --compile --install-headers /home/paul/env/images/include/site/python3.8/pyprismatic --enable-gpu
         cwd: /tmp/pip-install-takbc5vy/pyprismatic/
    Complete output (40 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.8
    creating build/lib.linux-x86_64-3.8/pyprismatic
    copying pyprismatic/fileio.py -> build/lib.linux-x86_64-3.8/pyprismatic
    copying pyprismatic/__init__.py -> build/lib.linux-x86_64-3.8/pyprismatic
    copying pyprismatic/params.py -> build/lib.linux-x86_64-3.8/pyprismatic
    running build_ext
    building 'pyprismatic.core' extension
    creating build/temp.linux-x86_64-3.8
    creating build/temp.linux-x86_64-3.8/pyprismatic
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DPRISMATIC_ENABLE_GPU=1 -DBUILDING_CUPRISMATIC=1 -I./include -I/home/paul/env/images/include -I/usr/include/python3.8 -c pyprismatic/core.cpp -o build/temp.linux-x86_64-3.8/pyprismatic/core.o -std=c++11
    In file included from ./include/params.h:21,
                     from pyprismatic/core.cpp:15:
    ./include/ArrayND.h: In member function ‘void Prismatic::ArrayND<N, T>::toMRC_f(const char*) const [with long unsigned int N = 3; T = std::vector<double>]’:
    ./include/ArrayND.h:389:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
      389 |    for (auto i = 0; i < this->size(); ++i)data_buffer[i] = (float)data[i];
          |                     ~~^~~~~~~~~~~~~~
    ./include/ArrayND.h: In member function ‘void Prismatic::ArrayND<N, T>::toMRC_f(const char*) const [with long unsigned int N = 3; T = std::vector<float>]’:
    ./include/ArrayND.h:414:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
      414 |    for (auto i = 0; i < this->size(); ++i)data_buffer[i] = data[i];
          |                     ~~^~~~~~~~~~~~~~
    ./include/ArrayND.h: In member function ‘void Prismatic::ArrayND<N, T>::toMRC_f(const char*) const [with long unsigned int N = 2; T = std::vector<double>]’:
    ./include/ArrayND.h:439:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
      439 |    for (auto i = 0; i < this->size(); ++i)data_buffer[i] = (float)data[i];
          |                     ~~^~~~~~~~~~~~~~
    ./include/ArrayND.h: In member function ‘void Prismatic::ArrayND<N, T>::toMRC_f(const char*) const [with long unsigned int N = 2; T = std::vector<float>]’:
    ./include/ArrayND.h:462:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
      462 |    for (auto i = 0; i < this->size(); ++i)data_buffer[i] = (float)data[i];
          |                     ~~^~~~~~~~~~~~~~
    ./include/ArrayND.h: In member function ‘void Prismatic::ArrayND<N, T>::toMRC_f(const char*) const [with long unsigned int N = 2; T = std::vector<unsigned int>]’:
    ./include/ArrayND.h:485:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
      485 |    for (auto i = 0; i < this->size(); ++i)data_buffer[i] = (float)data[i];
          |                     ~~^~~~~~~~~~~~~~
    x86_64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.8/pyprismatic/core.o -lfftw3f -lfftw3f_threads -lcuprismatic -lcufft -lcudart -o build/lib.linux-x86_64-3.8/pyprismatic/core.cpython-38-x86_64-linux-gnu.so
    /usr/bin/ld: cannot find -lcuprismatic
    collect2: error: ld returned 1 exit status
    error: command 'x86_64-linux-gnu-g++' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/paul/env/images/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-takbc5vy/pyprismatic/setup.py'"'"'; __file__='"'"'/tmp/pip-install-takbc5vy/pyprismatic/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-sugpug87/install-record.txt --single-version-externally-managed --compile --install-headers /home/paul/env/images/include/site/python3.8/pyprismatic --enable-gpu Check the logs for full command output.
```

I have the CUDa toolkit installed, and I've read the instructions about installing cuprismatic, however, the code to do so is commented out of prismatic version 1.2.1. How do I go about successfully installing the GPU version of pyprismatic, given that I can install the CPU version?

ericpre, 2021-07-21:
An alternative to build pyprismatic yourself is to use the conda-forge binary packages, which should work out of the box. Sadly, this is not documented and I have tried to improve the situation in https://github.com/prism-em/prism-em.github.io/pull/8, but unfortunately, this PR hasn't received any attention. In the mean time, you can have a look at the PR https://github.com/prism-em/prism-em.github.io/pull/8 itself for detailed instructions.

As an alternative, the cmake config have been improved in https://github.com/prism-em/prismatic/pull/76, so maybe you can have a look at it to get inspired to work around building issues of the release version.

drpaulsharp, 2021-08-31:
Installing the conda-forge binary packages has worked, but I needed to create a fresh conda environment for it to install correctly.

closed 2021-08-31 by drpaulsharp